makerdiary/nrf52840-mdk-usb-dongle #1: I don't understand the on board RGB led

RichardFerraro, 2018-12-10:
I am able to program the board just fine.

I do not understand the on board leds at P0.22 P0.23 P0.24 (I suspect it is 0.24)

Are they LEDS_ACTIVE_STATE 1 ?

caizelin, 2018-12-11:
> I am able to program the board just fine.
> 
> I do not understand the on board leds at P0.22 P0.23 P0.24 (I suspect it is 0.24)
> 
> Are they LEDS_ACTIVE_STATE 1 ?

@RichardFerraro Here is the schematic of RGB LED. It means that GREEN LED is connected to P0.22, RED LED is connected to P0.23 and BLUE LED is connected to P0.24. The `LEDS_ACTIVE_STATE` should be set to `0`.
[image]

caizelin, 2020-03-17:
I’m closing this issue because it has been inactive for a long time. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin