makerdiary/nrf52840-mdk-usb-dongle #4: nrf52840 not found in USB Device Tree MacOS Mojave v10.14

alnjxn, 2019-02-12:
Hello
Have just gone through the wpantund installation guide using Hombrew on MacOS. Have the service started as a dameon:

```
Name           Status  User Plist
dbus           started root /Users/ajackson/Library/LaunchAgents/homebrew.mxcl.dbus.plist
```

Plugged the USB device into two different USB hubs and neither show the device in the device tree. 
[image]

There are no LED's on either of the two USB dongles i purchased. When plugged into Mac. However when I plug into RaspberryPi (Plan to use 2nd device for border router) I at least get a blue LED.

caizelin, 2019-02-13:
@alnjxn When plugged into Mac, was the USB power indicator(Green LED on the dongle) lighting up?
Pls check the connection to ensure if the dongle is powered up correctly.

alnjxn, 2019-02-13:
@caizelin seems that the dongle was not seated properly in the hub. Green light is on and I can now see the device in the tree. Sorry for not properly checking that prior to posting. Thanks for the quick reply.

closed 2019-02-13 by alnjxn