makerdiary/nrf52840-mdk-usb-dongle #5: No bootloader access after flashing SDK app

mame82, 2019-02-13:
Hi,

I accidently flashed the `ble_app_blinky` for the 52840 dongle from Nordic (pca10059) before I noticed that this repo with custom board header files exist.

The dongle is still working and the BLE example app could be accessed via Bluetooth (of course the LED doesn't work, due to different PIN layout).

Anyways, pushing the button now, doesn't reset the MDK dongle back into bootloader mode (it has no effect at all). The bootloader hasn't been erased. I noticed that the Makefile for this board has `CONFIG_GPIO_AS_PINRESET` option disabled, which wasn't the case for the image compiled for pca10059, but I'm not sure if this is related.

I hope you can help in this, would be to bad if I bricked the dongle after flashing the first app.

caizelin, 2019-02-14:
@mame82 
Sorry for the late reply!
For nRF52840-MDK USB Dongle, you MUST undefine the CONFIG_GPIO_AS_PINRESET flag, or the Dongle would not enter the bootloader mode next time by holding the dongle's RESET/USR button.
pca10059 does NOT use P0.18 as GPIO, so it keeps the `CONFIG_GPIO_AS_PINRESET` option.

Do you have a debugger probe like JLink or DAPLink? Now you need a debugger probe for recovery.

BTW, I am considering to update the bootloader firmware with adding P4 as an alternate pin for entering bootloader mode. 

Thanks for your feedback!

mame82, 2019-02-14:
Thanks for fast reply.

Haven't a debugger probe yet, but will order one right away, if you could give me guidance on how to recover the bootloader.

Having an alternative PIN for bootloader entry would be awesome, btw

caizelin, 2019-02-14:
> Thanks for fast reply.
> 
> Haven't a debugger probe yet, but will order one right away, if you could give me guidance on how to recover the bootloader.
> 
> Having an alternative PIN for bootloader entry would be awesome, btw

@mame82 I am here to help you! I will also update the section to wiki page.

mame82, 2019-02-15:
Ordered this:

https://www.amazon.de/Adafruit-IBDAP-CMSIS-DAP-Adapter-ADA2764/dp/B01BT4N40S

Could it be used?

caizelin, 2019-02-15:
> Ordered this:
> 
> https://www.amazon.de/Adafruit-IBDAP-CMSIS-DAP-Adapter-ADA2764/dp/B01BT4N40S
> 
> Could it be used?

@mame82 , I have just checked that Adafruit-IBDAP-CMSIS-DAP-Adapter supports nRF51822 not nRF52840.
You can select a [SEGGER J-Link EDU Mini debugger](https://www.adafruit.com/product/3571). It is cheaper and and supports more targets.

mame82, 2019-02-15:
Too bad. Okay, placed an additional order on the SEGGER edu mini. Will reach back when it arrived

caizelin, 2019-02-15:
@mame82 Once you get the probe, just erase the chip of the dongle and then flash the bootloader firmware: [nrf52840_usb_dongle_open_bootloader_v1_1_0.hex](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/open_bootloader). 
Next time you compile a new sdk example, remember to undefine the `CONFIG_GPIO_AS_PINRESET` option(just comment `CFLAGS += -DCONFIG_GPIO_AS_PINRESET`).

mame82, 2019-02-21:
Managed to re-flash the bootloader with IBDAP CMSIS.

Ref on Twitter https://twitter.com/mame82/status/1098516205730779136?s=19

Any progress on bootloader with GPIO reset PIN ?

caizelin, 2019-02-21:
> Managed to re-flash the bootloader with IBDAP CMSIS.
> 
> Ref on Twitter https://twitter.com/mame82/status/1098516205730779136?s=19
> 
> Any progress on bootloader with GPIO reset PIN ?

Nice! New bootloader is working in progress for compatibility with the old one.

mame82, 2019-02-21:
Okay, so I think we could close. New bootloader isn't really off-topic, but better placed in the Wiki.

Agreed?

mame82, 2019-02-21:
... maybe there should be a note, that IBDAP CMSIS could be used for bootloader flashing with OpenOCD

caizelin, 2019-02-21:
> ... maybe there should be a note, that IBDAP CMSIS could be used for bootloader flashing with OpenOCD

You are right. I will update the docs for this. Thanks for your contributions!

caizelin, 2020-03-17:
I’m closing this issue because it has been inactive for a long time. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin