makerdiary/nrf52840-mdk-usb-dongle #6: USB generic HID with button events

mame82, 2019-02-17:
I finally got the usbd_generic_hid example of sdk 15.2 running. On both, the mdk dongle and the original nordic dongle.

Beside disabling CLI, I modified the example to only work with a single GPIO button (assigned to mouse left click).
USB enumeration works without issues now (HID mouse is detected by host).
The button related code in the example is based on bsp_events. In order to make this work, I undefined `bsp_simple` which is always enabled for the mdk dongle, unless a soft device is used, otherwise.

In contrast to the nordic dongle, no events are fired when the button is pressed.

Could you please explain how to make bsp_events work, without a softdevice.

I could provide the code including the modified makefile, if needed.

Thx in advance

caizelin, 2019-02-17:
> I finally got the usbd_generic_hid example of sdk 15.2 running. On both, the mdk dongle and the original nordic dongle.
>
> Beside disabling CLI, I modified the example to only work with a single GPIO [...]

@mame82 I just pushed a usbd_hid_generic example [here](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/examples/nrf5-sdk/usbd_hid_generic). I have tested that it works on the mdk dongle.
For your situation, two things you need to config:
1. Set `BTN_MOUSE_LEFT` to `0`, means that the button on the dongle will be used.
2. Comment `INIT_BSP_ASSIGN_RELEASE_ACTION(BTN_MOUSE_RIGHT);`, or the bsp will be failed to init.
That's it! You can try again.

mame82, 2019-02-17:
Weird, pretty much the changes I had done myself. Will give you example a shot tomorrow.

Thx for the quick update to the repo

caizelin, 2020-03-17:
I’m closing this issue because it has been fixed. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin