makerdiary/nrf52840-mdk-usb-dongle #7: Compiling the FTD CLI example with NRF5 SDK for Thread and Zigbee 3.0.0

cgawron, 2019-05-20:
After struggling with the Makefile and the sdk_config.h, I managed to compile the FTD CLI example.
However, the mdk usb dongle does not work with the image I created (green LED is on, but device does not show up in lsusb / dmesg).
The device works properly when I flash the hex file provided with the example, so I guess my build is broken.

Could you please update the `Makefile` and `sdk_config.h`, add an *ld config file* (I crafted one myself - that could also be the issue), and specify which version / flavour of the Nordic SDK is needed to compile the example?

I want to use the FTD example application to build my own sensor application based on the dongle.
An example featuring MQTT-SN and GPIO/I2C/SPI would also be appreciated.

caizelin, 2019-05-20:
@cgawron 👍 Thanks for your feedback. Will be update to the `3.0.0` as soon as possible.

cgawron, 2019-05-20:
Thank you for the quick response! 👍
Which version of the SDK should I use with the current example code & Makefile?

Would it help if I provide my changes (at least it compiles with 3.0.0 - to get there, I had to add some libraries and settings in sdk_config.h)?
Do you have any idea regarding the reason the dongle does not show up in lsusb?

caizelin, 2019-05-21:
> Thank you for the quick response! 👍
> Which version of the SDK should I use with the current example code & Makefile?
>
> Would it help if I provide my changes (at least it compiles with 3.0.0 - to get [...]
@cgawron  The current example code & Makefile are based on `v2.0.0`. `v3.0.0` is recommended.
Which host OS are you using? I will double check that.

cgawron, 2019-05-21:
@caizelin I'm using Linux (Debian stretch). I added some debugging code (LED blinking) and came to the conclusion that the example code hangs on calling `thread_instance_init()`.

BTW: The FTD CLI example from the openthread repository works fine. I'm also a bit confused regarding the different APIs (i.e. `thread_instance_init()` vs. `otInstanceInitSingle()`) - which one would you recommend? Do you know why Nordic provides an own Thread API?

caizelin, 2019-05-21:
Hi Christian, for Debian stretch, I need some time to check that. For other Linux distributions(like *Ubuntu* and *Raspbian for Raspberry Pi*), it works fine.

BTW, we have just updated the OpenThread examples(CLI and NCP) with the *nRF5 SDK for Thread and Zigbee V3.0.0*. Pls check them out here: https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/examples/openthread

I recommend to use the Nordic SDK examples as a template, as it includes more features. You can build the openthread repository to generate the static libraries, and then add them to the Makefile of the Nordic SDK examples. That is the way we use in this project.

Hope it's helpful for you.

cgawron, 2019-05-21:
@caizelin Thank you for the update, I have tested it and it works fine!

closed 2019-05-21 by cgawron