makerdiary/nrf52840-mdk-usb-dongle #8: No access to bootloader

TorstenRobitzki, 2019-07-25:
Hi,
I have some (3) new nRF52840 MDK USB Dongle, with cases. When I connect them to my PC (OS/X 10.14.6) and start nRF Connect, I can select the connected device and get some device informations but I will not get the chance to flash them (looks like the nRF Connect is still busy to interrogate the connected device:

[image]

If I connect a PCA10059, nRF Connect can get all required informations from the device:

[image]

I'm using nRF Connect version 3.0.0. I have tried two of the 3 USB Dongles, both have this issue.

caizelin, 2019-07-25:
@TorstenRobitzki 
Hi,
To program the Dongle, it must be in bootloader mode. While holding the dongle's RESET/USR button, connect it to your computer. When the Dongle has entered the bootloader mode, the RGB LED pulses RED. The Dongle is now ready for programming.

You can also check out the programming tutorial here: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/

TorstenRobitzki, 2019-07-25:
Thanks, works like a charm!

closed 2019-07-25 by TorstenRobitzki