makerdiary/nrf52840-mdk-usb-dongle #9: Button immediately puts device into Bootloader mode

SriramKeerthi, 2019-07-27:
I have a couple of dongles, and they both seem to be flashing correctly. However on one of them, the device seems to put itself in bootloader mode any time I press the button, vs only when holding down the button before inserting it into the USB.

To verify that I didn't accidentally flash something that was causing this, I flashed the `ble_app_blinky_without_sd.hex` and the s140 soft device. Flashing worked and the device was recognized and I was able to change the LED using the nRF Connect app on the phone. I did the same thing with my second dongle. The first one still seems to enter bootloader on button press, but the second one behaves as expected - only entering bootloader when plugged in with the button down.

Do you think I flashed some example from the nRF SDK that might have overwritten the behavior of the button? Can I flash something over this? I tried rewriting the bootloader hex, but that doesn't seem to be signed so that's not working.

caizelin, 2019-07-27:
@SriramKeerthi 
The pin P0.18 connected to the button is configed as GPIO default. If you have flashed some example from the nRF5 SDK, the button will be changed to a RESET pin.

Pls check if you have undefined the marco CONFIG_GPIO_AS_PINRESET in your application's makefile. 

If you want to restore the default settings, you can erase the pselreset by flashing this example: https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/examples/nrf5-sdk/pselreset_erase

SriramKeerthi, 2019-07-27:
That worked perfectly! Thank you so much!

caizelin, 2020-03-17:
I’m closing this issue because it has been fixed. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin