makerdiary/nrf52840-mdk-usb-dongle #10: Debugging dongle with the nRF52840 PDK

SriramKeerthi, 2019-07-30:
I have a [Nordic nRF52840 Preview DK](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html), which is very similar to the nRF52840 DK available from Nordic Semiconductor.

I tried connecting the P20 on the DK to the dongle as follows:

```
DK              DONGLE
GND DETECT  ->  GND
RESET       ->  RST
SWD CLK     ->  SWDCLK
SWD IO      ->  SWDIO
```

Unfortunately the board doesn't seem to be recognizing that the device is connected and will not program it. Is my connection correct or am I missing something?

caizelin, 2019-08-02:
@SriramKeerthi Sorry for the late reply.
 How did you deal with the power of the dongle and DK?
Here is the schematic:
[image]

You can also follow this topic from Nordic DevZone:
https://devzone.nordicsemi.com/f/nordic-q-a/32661/external-programming-with-nrf5-dk-p20

SriramKeerthi, 2019-08-03:
Thanks! The board I have is the PDK, which is a bit different from the DK (Has a few more pins and labelled a bit differently), I will report back as soon as it works.

SriramKeerthi, 2019-08-03:
@caizelin what is the VDD on the dongle? Is it VIN? Is RST unconnected?

caizelin, 2019-08-03:
> @caizelin what is the VDD on the dongle? Is it VIN? Is RST unconnected?

If the power is 3.3V, connect to `3V3` on the dongle. 
If it is 5V, use `VIN` instead.
The RST can be left unconnected.

SriramKeerthi, 2019-08-06:
Thanks a lot! I purchased a JLINK SWD debugger paralelly and managed to get this to work by connecting:
```
JLINK           DONGLE
GND         ->  GND
VCC         ->  3V3
SWD CLK     ->  SWDCLK
SWD IO      ->  SWDIO
```

closed 2019-08-06 by SriramKeerthi