makerdiary/nrf52840-mdk-usb-dongle #11: Antena

y0d4a, 2019-08-20:
Hi, i know this is not issue, but did not found better place to ask this :)
is it possible to increase wireless signal on this dongle?

thank you.

y0d4a, 2019-08-20:
for ex ample, to switch to similar one https://www.aliexpress.com/item/33007560614.html?spm=2114.12057483.0.0.289241a7a740c5

caizelin, 2019-08-20:
> Hi, i know this is not issue, but did not found better place to ask this :)
> is it possible to increase wireless signal on this dongle?
> i see that there is antenna as chip, is it possible to unsolder and solder something stronger?
> 
> thank you.

Hi there,
Have you used the internal PA? For example, set the tx power using:
```
 APP_ERROR_CHECK(sd_ble_gap_tx_power_set(BLE_GAP_TX_POWER_ROLE_ADV, 0, 8));
```

emenezes, 2019-08-31:
Hi @caizelin, I'm using your usb-dongle to run some experiments for my thesis. I did not find a BOM (bill of materials) in your folders, so I would like to know if you may disclose the make and model of the antenna chip. It would be very interesting for me to have the documentation and radiation pattern of it.

thanks in advance,
Great work.

caizelin, 2019-09-01:
> Hi @caizelin, I'm using your usb-dongle to run some experiments for my thesis. I did not find a BOM (bill of materials) in your folders, so I would like to know if you may disclose the make and model of the antenna chip. It would be very interesting for me to have its documentation and radiation pattern diagram.
> 
> thanks in advance,
> Great work.

@emenezes Hi, the chip antenna is ACA-2012-A1-CC-S. You can find the spec here: http://www.inpaqgp.com/rw_40b272d49797af2cc6cacf409f1a9df0

emenezes, 2019-09-03:
Thank you very much for the info, I will look into it.

caizelin, 2020-03-17:
I’m closing this issue because it has been inactive for a long time. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin