makerdiary/nrf52840-mdk-usb-dongle #12: Is there a forum for the nrf52840-mdk-usb-dongle?

hpssjellis, 2019-08-25:
Wondering if there is a forum for the nrf52840-mdk-usb-dongle?

I teach a Robotics course using the https://Particle.io Photon and am curious about doing similar work using the nrf52840 dongle. All my course work and videos are at this Github site:
https://github.com/hpssjellis/particle.io-photon-high-school-robotics

Getting some of these sensors and actuators working on the dongle would be big job. Wondering if anyone is interested. We could just keep this thread open. Any suggestions?

Jeremy Ellis  https://twitter.com/rocksetta

caizelin, 2019-08-28:
> Wondering if there is a forum for the nrf52840-mdk-usb-dongle?
>
> I teach a Robotics course using the https://Particle.io Photon and am curious about doing similar work using the nrf52840 dongle. All [...]

@hpssjellis Yes, we have a forum under construction: https://community.makerdiary.com/

caizelin, 2020-03-17:
I’m closing this issue because it has been inactive for a long time. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin