makerdiary/nrf52840-mdk-usb-dongle #13: Dongle is zigbee compatible ?

faronov, 2019-09-06:
Hi I buy this dongle and try to use nRF5 Zigbee SDK
I'm able successfully to build and flash examples
but when it starts ZigBee part it always freezes
I didn't find any information, about ZigBee compatibility on your site

Does this mean that dongle not compatible with Nordic Zigbee SDK?

caizelin, 2019-09-06:
@faronov Hi, it should be compatible with nRF5 Zigbee SDK as this dongle is based on the same nRF52840 chip. Our ZigBee examples are under construction and currently you can refer to [Nordic's Zigbee SDK](https://infocenter.nordicsemi.com/topic/sdk_tz_v3.1.0/zigbee_main.html) docs for more details.

faronov, 2019-09-06:
I compile ZigBee app cli example
when I start it as a router or coordinator with command 
```
bdb role zc
bdb start
```
the last line says that the coordinator is started
after that module is not responded

faronov, 2019-09-06:
I use PCA10059, maybe I must change something in sdk_config for compatibility?

uKev, 2022-12-17:
Is there any update on this?

On Amazon this product is advertised as:

"Features
Low cost USB dongle supporting Bluetooth LE, Thread, **Zigbee**, 802.15.4, ANT and 2.4GHz
Pre-built OpenThread NCP Firmware support
OpenThread Border Router support
Thread Network Sniffer with Wireshark"

Ping @caizelin please clarify

caizelin, 2022-12-18:
> Is there any update on this?
>
> On Amazon this product is advertised as:
>
> "Features Low cost USB dongle supporting Bluetooth LE, Thread, **Zigbee**, 802.15.4, ANT and 2.4GHz Pre-built OpenThread NCP [...]

@uKev Yes, the nRF52840 supports Zigbee. You can find the examples in nRF Connect SDK or use the Zigbee SDK:
* [NCS Zigbee examples](https://github.com/nrfconnect/sdk-nrf/tree/main/samples/zigbee)
* [nRF5 SDK for Thread and Zigbee v4.2.0](https://infocenter.nordicsemi.com/topic/struct_sdk/struct/sdk_thread_zigbee_latest.html)

caizelin, 2023-08-09:
The latest Zigbee examples can be found here: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/guides/ncs/samples/zigbee/lighting/
I will close this issue. If still have any questions, please reopen it.

closed 2023-08-09 by caizelin