makerdiary/nrf52840-mdk-usb-dongle #14: User button doesn't work

ananasn, 2019-10-29:
I've struggled to get user button on the dongle to work for the last few days. Still no success. I took the blinky example and modify it a little bit with gpio calls. Looks like the statement nrf_gpio_pin_read(BUTTON_1) in the code always equals 1 regardless I push button or not. What am I doing wrong?
```
#include <stdbool.h>
#include <stdint.h>
#include "boards.h"
#include "nrf_delay.h"

int main(void)
{
    nrf_gpio_cfg_output(LED_3);
    nrf_gpio_cfg_input(BUTTON_1, BUTTON_PULL);
    for (;;) {
        if (nrf_gpio_pin_read(BUTTON_1) == 1) {
            nrf_gpio_pin_toggle(LED_3);
            nrf_delay_ms(200);
        }
        else {
            nrf_gpio_pin_write(LED_3, 1);
        }
    }
}
```

aeozyalcin, 2020-03-01:
I am also confused by whether the single button on the 52840 USB dongle is actually a user programmable button, or if it is purely reset/bootloader button. I haven't been able to get the button working. I hope @caizelin can respond!

caizelin, 2020-03-01:
> I am also confused by whether the single button on the 52840 USB dongle is actually a user programmable button, or if it is purely reset/bootloader button. I haven't been able to get the button working. I hope @caizelin can respond!

Hi,
The pin P0.18 connected to the button is configed as GPIO default. If you have flashed some example from the nRF5 SDK, the button may be changed to a RESET pin.

Pls check if you have undefined the marco CONFIG_GPIO_AS_PINRESET in your application's makefile.

If you want to restore the default settings, you can erase the pselreset by flashing this example: https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/examples/nrf5-sdk/pselreset_erase

aeozyalcin, 2020-03-02:
> Hi,
> The pin P0.18 connected to the button is configed as GPIO default. If you have flashed some example from the nRF5 SDK, the button may be changed to a RESET pin.
>
> Pls check if you have undefined [...]

That did the trick! Thank you!

Is there a reason why the MDK52840 USB dongle version of the MDK has its examples based on the 15.0 version of the SDK, while the regular MDK52840 has it based on the 15.2 version? Is there a plan to update the MDKs so they use the latest 16.0 version of the Nordic SDK?

caizelin, 2020-03-02:
@aeozyalcin Yes. These boards should all work with 16.0, and we will update the wiki as soon as possible. `16.0` will be the last version of the Nordic SDK. After that, Nordic will replace the nRF5 SDK with the nRF Connect SDK(NCS).

ananasn, 2020-03-08:
> Hi,
> The pin P0.18 connected to the button is configed as GPIO default. If you have flashed some example from the nRF5 SDK, the button may be changed to a RESET pin.
>
> Pls check if you have undefined [...]

Hi, 
I just checked the CONFIG_GPIO_AS_PINRESET flag, it is undefined in Makefile.common:
```
#CFLAGS += -DCONFIG_GPIO_AS_PINRESET
```
I also flashed the firmware you gave a link above, just in case.  And then flashed my program from the topic. However, still no success.

Could you give me a workable example of the user button applying ```nrf_gpio_pin_read``` function?

mischief, 2020-07-05:
@ananasn hi. i found that the default button definition didn't work for me when i tried to run `ble_app_blinky`. i changed the button definition to match what is in the nrf52840 dongle schematic (P0.18) and then i could read the button state with the example.

```
diff --git a/examples/nrf5-sdk/ble_app_blinky/config/custom_board.h b/examples/nrf5-sdk/ble_app_blinky/config/custom_board.h
index c8ccd42..5d1ec74 100755
--- a/examples/nrf5-sdk/ble_app_blinky/config/custom_board.h
+++ b/examples/nrf5-sdk/ble_app_blinky/config/custom_board.h
@@ -67,7 +67,7 @@ extern "C" {
 
 #define BUTTONS_NUMBER 1
 
-#define BUTTON_1       NRF_GPIO_PIN_MAP(1,0)
+#define BUTTON_1       NRF_GPIO_PIN_MAP(0,18)
 #define BUTTON_PULL    NRF_GPIO_PIN_PULLUP
 
 #define BUTTONS_ACTIVE_STATE 0
```

@caizelin is there a reason the button definition in the repo is not P0.18?

caizelin, 2023-08-09:
You can try this example to test the button: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/guides/ncs/samples/button/
This example will re-configure the button as GPIO functionality if it was set to RESET before.
Will close for now, and if you see it again, please consider re-opening it.

closed 2023-08-09 by caizelin