makerdiary/nrf52840-mdk-usb-dongle #16: OpenSK compatible?

tmikaeld, 2020-01-31:
Would this work with https://github.com/google/OpenSK ?

caizelin, 2020-01-31:
> Would this work with https://github.com/google/OpenSK ?
> 
> EDIT: Also mentioned here [google/OpenSK#3](https://github.com/google/OpenSK/issues/3)

Hi there,
You are really up-to-date! Yes, the hardware is compatible and we are working in process to adapt the LEDs and button. @kfihihc @xiongyihui

tmikaeld, 2020-01-31:
Hello @caizelin!

Superb! I can't wait to order these devices and try it out.

Thanks for doing this, FIDO2 really needs to get out there :)

xiongyihui, 2020-02-04:
@tmikaeld You can follow [Get Started with OpenSK guide](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/opensk/getting-started/) to use FIDO2 with nRF52840 MDK USB Dongle

Raboo, 2020-02-04:
Great news!
I installed OpenSK according to your *Getting Started with OpenSK* guide and wanted to report my results.
Currently it only partially works. It works on my MacOS device (which I used to install OpenSK). It did not work on my Windows 10 device nor on my Android device (using same USB-C to USB dongle I used on Mac).

Also am I correct that the dongle has a antenna that can be used with NFC? I also tried NFC with my phone, that didn't work either, it did not find any NFC device.

xiongyihui, 2020-02-05:
@Raboo It's weird.  It works on my Windows 10 (version 1909)

The dongle has two NFC pins, but NFC is not enabled at OpenSK yet

tmikaeld, 2020-02-05:
@Raboo Should work in latest Win10, I've had issues with older version when using Yubikey as well. Also, Windows Hello is super-buggy in general....

Raboo, 2020-02-05:
I have Windows 10 Pro, Version 1909.
https://webauthn.io/ looks like this when trying to login (registering seems to work).
[image]

I also tested https://demo.yubico.com/webauthn-technical/registration and I wasn't able to login from my registration that I did on my Mac, I had to register again, then it worked. I don't know how long the registration is valid. It went some hours between register on Mac and login on Windows.

tmikaeld, 2020-02-05:
These tests don't use any server, it only stores the registration locally so you can test it. So re-registration is always needed.

Raboo, 2020-02-05:
I tried again, with Firefox, Edge, Chrome. All of them have problems on webauthn.io.
So I need to register the security key somehow?

Raboo, 2020-02-06:
> The dongle has two NFC pins, but NFC is not enabled at OpenSK yet

A bit off topic here... Maybe I should move this question to a new issue?
I'm a bit novice when it comes to this. Those pins need to be connected to the onboard antenna or I need to buy a NFC antenna and solder it?
Also would a battery or a capacitor be required for this then, like mentioned here https://github.com/google/OpenSK/issues/15#issuecomment-582645999?

tmikaeld, 2020-02-07:
I'll close this issue, since it's now clearly compatible and working 👍

Thanks @xiongyihui for looking at it so quickly!

closed 2020-02-07 by tmikaeld

tmikaeld, 2020-02-29:
Just got the device and programmed it in a few easy steps, works flawlessly 👍