makerdiary/nrf52840-mdk-usb-dongle #17: flashing dongle

pabloopez, 2020-02-05:
I have been in the openSK documentation on GitHub that there is need for an special propietary device to flash de dongle. Is that right? Isn't it possible to flash it by usb? Thanks!

caizelin, 2020-02-05:
> I have been in the openSK documentation on GitHub that there is need for an special propietary device to flash de dongle. Is that right? Isn't it possible to flash it by usb? Thanks!

Hi,
Our nRF52840 MDK USB Dongle contains a bootloader, so it can be flashed via usb and no other special propietary device needed.
Please follow this guide to getting started: [Getting Started with OpenSK](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/opensk/getting-started/)

caizelin, 2020-03-17:
I’m closing this issue because it has been fixed. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin