makerdiary/nrf52840-mdk-usb-dongle #18: nrf52840 dongle issue--Nrf Connect does not support

dharminec, 2020-02-09:
[image]

caizelin, 2020-03-17:
@dharminec The connectivity example in nRF5 SDK does not work with pc-nrfconnect-ble app. You should compile the connectivity firmware from https://github.com/NordicSemiconductor/pc-ble-driver. Because some modifications are needed to patch the connectivity example in nRF5 SDK.

Here are the pre-compiled firmwares you can have a try:
https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/ble_connectivity

caizelin, 2020-03-17:
I’m closing this issue because it has been fixed. Please reopen if you still encounter this issue.
Thank you!

closed 2020-03-17 by caizelin