makerdiary/nrf52840-mdk-usb-dongle #19: NFC + OpenSK

Raboo, 2020-02-11:
I'm a bit novice when it comes to this, but what is required to get NFC to work with OpenSK on this nRF52840-mdk-usb-dongle?
Does a NFC antenna need to be soldered on the board?
Would a battery or a capacitor be required for this then, like mentioned here https://github.com/google/OpenSK/issues/15#issuecomment-582645999?

Or is it just the lack of NFC support in OpenSK?

tmikaeld, 2020-02-29:
The dongle doesn't have NFC working unfortunately.
I'm also curious if we can add an antenna ourselves? 

NFC doesn't require a battery, the Yubikey works with NFC and it doesn't have any battery whatsoever.

CuBiC3D, 2020-03-01:
@tmikaeld You can add an antenna for NFC, the pins are routed to the side connector on PCB. But it still requires a power source, since this particular SoC can not run from the power of the NFC field. The Yubikeys with NFC support are very low power devices.

I am also interested on the specifics for the antenna (form, dimensions, caps). Would be very interesting to have a reference modification.

AngainorDev, 2020-10-24:
Interested as well.