makerdiary/nrf52840-mdk-usb-dongle #20: Where to find uf2conv.py

abuckton, 2020-02-27:
I have a new MDK that came with the UF2 boot loader (took me a while to work that out ... there really should be documentation to say that it's changed).

When I look how to do DFU using UF2 (https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/uf2_bootloader) it mentioned uf2conv.py.
Where is that script? (It is not on the link "[Python Conversion Script](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/tools)" 

Thanks,
Anthony

abuckton, 2020-02-27:
As an interim, I've found the Microsoft version of the script: https://github.com/Microsoft/uf2/blob/master/utils/uf2conv.py and pasted it into /usr/local/bin
 It seemed to work as I now have my Thread boarder router talking to wpantund and wpanctl

caizelin, 2020-03-01:
> As an interim, I've found the Microsoft version of the script: https://github.com/Microsoft/uf2/blob/master/utils/uf2conv.py and pasted it into /usr/local/bin
> It seemed to work as I now have my Thread boarder router talking to wpantund and wpanctl

@abuckton That's okay! The script is ignored by the `.gitignore` file, and I have just fixed it. This script is also hosted in the [tools](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/tools) directory.

commit 437270fd (2020-03-01): Fixes #20
closed 2020-03-01 by caizelin