makerdiary/nrf52840-mdk-usb-dongle #22: Blinky .hex does not work

oschade, 2020-03-08:
Hello,

i can't make the blinky example run on my nrf52840 mdk usb dongle.
I followed the tutorial at:

https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/nrf5-sdk/
and
https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/

I compiled one blinky example by myself, but neither this one nor the one from this repository can be transfer to the usb dongle. Please see the picture below:

[image]

Where is the problem? I was able to transfer a soft device and the ble blinky example. But this doesn't show any ble signal in my smartphone. What am I doing wrong?

caizelin, 2020-03-17:
@oschade Please check the `blinky_gcc_nrf52.ld` file to make sure using the right memory regions:
```
MEMORY
{
  FLASH (rx) : ORIGIN = 0x1000, LENGTH = 0xff000
  RAM (rwx) :  ORIGIN = 0x20000008, LENGTH = 0x3fff8
}
```

oschade, 2020-03-27:
Cool, now it works! Thank you!

closed 2020-03-27 by oschade