makerdiary/nrf52840-mdk-usb-dongle #23: Unable to flash firmware after upgrading bootloader to UF2

BenBergman, 2020-03-11:
I was having some issues reliably pushing new .hex files via OpenBootloader on one of my computers so I decided to try switching to the UF2 bootloader (using another computer that has been more reliable).

After performing the upgrade, I did see the storage device and was able to upload the blink program (I generated the .uf2 file from the .hex).

Prior to the upgrade, any time I hit the reset button (either holding it before insert or pressing it after insert) would put it into bootloader mode. Now pressing the button blinks the red light briefly and then goes back to the blinky example. The same happens if I press and hold while inserting. The UF2 drive will not show up in either case.

BenBergman, 2020-03-12:
Some additional info: Prior to switching to the UF2 bootloader I tried installing the latest OpenBootloader and nrfutil gave me an error saying the version of the new bootloader needed to be greater than the old version, so I assume my board already had the latest OpenBootloader on it previously.

BenBergman, 2020-03-12:
Oh, one more thing. The machine that was not flashing well is running up to date Arch Linux. I tried initially through some USB hub/extenders mounted to my desk (both USB 2 and USB 3) and also tried directly connected to the PC itself. all showed similar unreliability.

On the machine that WAS able to flash firmware and which was used to update the bootloader, it is a MacBook Pro running macOS 10.14.6.

BenBergman, 2020-03-12:
Since the firmware is based on Adafruit's UF2 bootloader, I decided to try double clicking the button. After blinking red for a second or two the light goes solid green and the drive shows up and I'm able to program it.

It seems I can also get to the bootloader by holding the button while inserting and then quickly pressing it a second time immediately after releasing it.

The documentation for the UF2 bootloader usage should be updated to reflect this behaviour. It is also unclear how to load a soft device with your application code in a UF2.

caizelin, 2020-03-17:
@BenBergman Thanks for your feedback!
The pin P0.18 connected to the button is configed as GPIO default.  If you flashed the blinky example with `CONFIG_GPIO_AS_PINRESET` defined, the button will be used as RESET function NOT a GPIO. In this case, you have to double-click the button to enter the UF2 bootloader.

To enter the bootloader by holding the button while inserting, the button must be configed as the GPIO function(the default setting). 

You can erase the `pselreset` to  config button as GPIO by flashing this example: https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/examples/nrf5-sdk/pselreset_erase .

For loading a firmware with softdevice, just merge the softdevice with app, and then convert the merged hex into the UF2.

BenBergman, 2020-03-18:
Thanks for the info @caizelin ! I'll give that a try when I get a chance.

caizelin, 2023-08-09:
Will close for now, and if you see it again, please consider re-opening it.

closed 2023-08-09 by caizelin