makerdiary/nrf52840-mdk-usb-dongle #24: How to connect to Jtag ?

ToutUnProg, 2020-03-20:
Hi,

I wanted to flash the firmware to put TockOS into the nrf52840-mdk-usb-dongle by using a JTag probe.
I build a card to connect the usb jtag probe to the dongle and connect everything as it is in the attached scheme.
[image]

Is there something wrong ? can you help me please.

caizelin, 2020-03-20:
@ToutUnProg It's right. You can do this.

ToutUnProg, 2020-03-21:
OK, I did this on a little Card. But it does not work.
Is there a process to power up the JTag and the dongle that I cannot find ?
[image]
[image]
[image]

caizelin, 2023-08-09:
See this guide that may help: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/hardware/#debug-interface
Will close for now, and if you still have any questions, please consider re-opening it.

closed 2023-08-09 by caizelin