makerdiary/nrf52840-mdk-usb-dongle #25: nRF52840 MDK USB Dongle question

Brechen, 2020-03-27:
Hello, can I use this device directly on my Mac? I want to use it as an Ant+ transmitter. Do I have to program it or can be used on the go? Many thanks

caizelin, 2020-05-07:
> Hello, can I use this device directly on my Mac? I want to use it as an Ant+ transmitter. Do I have to program it or can be used on the go? Many thanks

@Brechen You can use this device on macos. Here is the guide you can follow to program this dongle: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/

caizelin, 2023-08-09:
There was no response for a long time. Will close for now, and if you still have trouble with it, please consider re-opening this issue.

closed 2023-08-09 by caizelin