makerdiary/nrf52840-mdk-usb-dongle #27: Not able to copy (CircuitPython) UF2 files to MDK drive

Shark009, 2020-04-28:
Hi!

On OS X I'm unable to copy the CircuitPython UF2 file to the MDK drive. It disconnects automatically as soon as I start copying the file. I've used UF2CONV to convert the HEX file to UF2 . I also tried to use the nRF connect tools to get the hex file loaded that way. However that did't detect the board. So both options didn't work.

Are there some pointers to get further?

theBASTI0N, 2020-06-01:
I also tried this and now my MDK drive is not working at all.

gbuzogany, 2020-07-23:
AFAIK, to use circuitpython, you should flash the Adafruit nRF52 bootloader:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases

The mdk_nrf52840_dongle_* should be what you are looking for.

After flashing the bootloader, you should be able to flash circuitpython (https://circuitpython.org/board/makerdiary_nrf52840_mdk_usb_dongle/)

I hope this is helpful.

If you "bricked" your dongle (can't enter DFU), you can follow the instructions in this video (https://www.youtube.com/watch?v=R5wub5ywzTU) to flash it though the SWD interface using a raspberry pi.

caizelin, 2023-08-09:
Also refer to our programming guide: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/
There was no response for a long time. Will close for now, and if you still have trouble with it, please consider re-opening this issue.