makerdiary/nrf52840-mdk-usb-dongle #29: NRF52840 dongle执行无法执行python库

SYJshang, 2020-05-07:
```
在nrf connect desktop 可以识别并工作，但是我在跑官方给的pc-ble-driver-py example中 serial_port_enum.py例子时候报错。

$python3 serial_port_enum.py NRF52 tty.usbmodemC20FBBC0D20F2
[]
enum_serial_ports: 0 serial ports found
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/pc_ble_driver_py/ble_driver.py", line 2732, in __init__
    snr = [d.serial_number for d in serial_ports if d.port == serial_port][0]
IndexError: list index out of range

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "serial_port_enum.py", line 87, in <module>
    main(port)
  File "serial_port_enum.py", line 66, in main
    flasher = Flasher(serial_port=selected_port)
  File "/usr/local/lib/python3.7/site-packages/pc_ble_driver_py/ble_driver.py", line 2734, in __init__
    raise NordicSemiException("Board not found: {}".format(serial_port or snr))
pc_ble_driver_py.exceptions.NordicSemiException: Board not found: tty.usbmodemC20FBBC0D20F2

但是我是可以看到这个设备的
$ls tty.*
tty.Bluetooth-Incoming-Port tty.usbmodemC20FBBC0D20F2
```
这个问题要怎么处理呢？

caizelin, 2020-05-09:
@SYJshang Try the conectivity firmware with softdevice s132.5.1 here: https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/ble_connectivity

SYJshang, 2020-05-09:
你好，我使用5.1版本在这个dongle上是不成功的，在nrf connect不能使用。
[image]

caizelin, 2020-05-09:
nRF Connect app only supports s132_3.1.0 currently. If you want to use pc-ble-driver-py, the s132_5.1.0 is needed.

caizelin, 2023-08-09:
Please refer to the quick start guide now: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/getting-started