makerdiary/nrf52840-mdk-usb-dongle #31: thread sniffer

bashendriks123, 2020-05-14:
I'm new to this board/Python etc.
I bought this dongle in the hope to be able to sniff some bluetooth data on a windows environment.
Does this dongle also work on windows (10)?
I tried to get it to work but i do not get any data from the com port where the dongle is connected to using the thread sniffer approach. When using the python C:\nrf52840\pyspinel-master\sniffer.py -c 11 -n 1 -b 9600 --crc -u com4 --no-reset | "C:\Program Files\Wireshark\Wireshark.exe" -k -i -
i get wireshark to accept the com port but there is no data from this port . (in wireshark i get a - as a capture device)

When i use the C:\Program Files\Wireshark\extcap\extcap_ot.bat and extcap_ot.py approach i also do not get any errors but also not data and no capture device (or whatever) in wireshark.
the test:
C:\Program Files\Wireshark\extcap>extcap_ot.bat --extcap-interfaces
extcap {version=1.0.0}{display=OpenThread Sniffer}{help=https://github.com/openthread/pyspinel}
just gives one line (is that ok?)
When i use the nRF Connect (desktop tool) combined with the programmer then i get an errir after selecting the device : "Unsupported device. The detected device could not be recognized as neither JLink device nor Nordic USB device."
the device and the serial name are correct
This seems to be a jlink driver issue. Using different jlink drivers did not result in a solution... the error is still the same.
furthermore i tried to flash the dongle without any result since the dongle cannot be connected to.
Any suggestions how to get this to work?

gbuzogany, 2020-07-23:
AFAIK you can't use the NRF Connect Programmer with the UF2 bootloader. If you want to use the NRF Programmer, you should flash the Open Bootloader.

Also, you mentioned you want to make a bluetooth sniffer, but you are flashing a Thread network sniffer. Is it what you intend? Sniff Thread network packets?

bashendriks123, 2020-07-23:
I'm already Happy to at least sniff something.... Thx

gbuzogany, 2020-07-23:
The dongle should work on windows 10, but the guides aren't complete/up to date even for the platform they are written for (macos), so it will probably require some experimenting.

To use the thread sniffer, there is a guide for macos (https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/thread-sniffer/), but the guide assumes (even if it doesn't say so) that you already flashed the firmware/openthread/cli/*ftd* firmware to the dongle. Also, the guide and the firmwares are not up to date, so if you look for other thread resources, there will be some inconsistencies.

If you want a _bluetooth_ sniffer, you should check this repo:
https://github.com/makerdiary/raccoon

bashendriks123, 2020-07-24:
Thx will look in to this.

afuen010, 2023-01-09:
> https://github.com/makerdiary/raccoon

I tried to flash firmware/openthread/cli/ftd. I assume you mean this one https://github.com/makerdiary/nrf52840-mdk-usb-dongle/blob/master/firmware/openthread/cli/thread_cli_ftd_nrf52840_mdk_usb_dongle_v1.3.0.zip. When I tried to flash it with nrfutil or with adafruit-nrfutil. I got nothing. Is this the right one?

caizelin, 2023-08-09:
Solved here: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/guides/nrf802154-sniffer/

closed 2023-08-09 by caizelin