makerdiary/nrf52840-mdk-usb-dongle #32: Can not enter bootloader mode after flashed the newest OpenSK firmare via UF2 bootloader

snowyu, 2020-05-20:
```bash
nrfutil dfu usb-serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p /dev/ttyACM0
git clone https://github.com/google/OpenSK
cd OpenSK
./setup.sh
./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none
python ../uf2conv.py -c -f 0xada52840 -o mdk_dongle_opensk_all_in_one.uf2 target/nrf52840_mdk_dfu_merged.hex
```
The OpenSK work well, but it can not enter bootload mode (hold on the button before inserting) after copy `mdk_dongle_opensk_all_in_one.uf2` image to MDK-DONGLE.

mtron, 2020-05-31:
Hello same problem here. 

https://github.com/google/OpenSK at git revision: af79d92a888fb398bd1e630f1d821c4ed4128f95
( Wed May 13 17:48:23 2020 +0200)

Flashed with: 
./deploy.py --board nrf52840_mdk_dfu --opensk --programmer nordicdfu

OpenSK works but it is not possible to enter dfu mode. Any way to reset the stick?

gbuzogany, 2020-07-23:
I made a few mistakes and had the same issue: could not put the stick in DFU mode.

The best solution I found was to flash it through SWD using OpenOCD on a raspberry pi. This video was super helpful: https://www.youtube.com/watch?v=R5wub5ywzTU

caizelin, 2023-08-09:
Refer to our OpenSK guide here: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/guides/opensk/

closed 2023-08-09 by caizelin