makerdiary/nrf52840-mdk-usb-dongle #33: Followed build process for OpenSK, encounterd error

Prosthet1cHead, 2020-06-17:
Following the instructions here:
https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/opensk/building/

Everything appears to go well until the "board=nrf52840_mdk_usb_dongle ./deploy.sh app" stage, which dies as shown below.

```
root@Laptop:~/OpenSK# board=nrf52840_mdk_usb_dongle ./deploy.sh app
info: syncing channel updates for 'nightly-2020-01-16-x86_64-unknown-linux-gnu'

  nightly-2020-01-16-x86_64-unknown-linux-gnu unchanged - rustc 1.42.0-nightly (3291ae339 2020-01-15)

info: checking for self-updates
info: component 'rust-std' for target 'thumbv7em-none-eabi' is up to date
./deploy.sh: line 303: tockloader: command not found
   Compiling byteorder v1.3.4
   Compiling linked_list_allocator v0.6.6
error[E0432]: unresolved import `alloc::alloc::AllocRef`
  --> /root/.cargo/registry/src/github.com-1ecc6299db9ec823/linked_list_allocator-0.6.6/src/lib.rs:14:20
   |
14 | use alloc::alloc::{AllocRef, AllocErr, Layout};
   |                    ^^^^^^^^ no `AllocRef` in `alloc`

error: aborting due to previous error

For more information about this error, try `rustc --explain E0432`.
error: could not compile `linked_list_allocator`.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

I have zero rust language experience so can't debug this myself.
Running the process as root on Ubuntu Mate 20.04.

pbchekin, 2020-10-18:
I have the same issue

pbchekin, 2020-10-18:
@xiongyihui could you please take a look? In the OpenSK upstream they fixed it with nightly-2020-02-03.

xiongyihui, 2020-10-19:
The document is find of outdated.
We can follow https://github.com/google/OpenSK/blob/master/docs/install.md to build the firmware.

```
./setup.sh
./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none
```
It will create a hex at the `target` directory.

xiongyihui, 2020-10-19:
Please use https://github.com/makerdiary/OpenSK/tree/master which includes a commit to avoid breaking the bootloader.

alishir, 2021-05-03:
@xiongyihui I've downloaded the `deploy.py` script from google/OpenSK and run the following command:
```bash
./setup.sh
./deploy.py --board=nrf52840_mdk_dfu --opensk --programmer=none
```
But I got this error:
```
deploy.py: error: argument --board: invalid choice: 'nrf52840_mdk_dfu' (choose from )
```