makerdiary/nrf52840-mdk-usb-dongle #34: I have absolutely no idea what state my dongles are in / UF2 confusion / old documentation

ctshh, 2020-07-07:
I have bought a bunch of dongles, the come up in a state in which the nRF tool sees the serial port, yet cannot use them (`"Unsupported device. The detected device could not be recognized as neither JLink device nor Nordic USB device."`) and holding the button on power-up yields a UF2 flash drive, yet it is not possible to upgrade to a (correctly hex-to-UF2 converted) CircuitPython.

Please urgently update the documentation and give a quick explanation here on how to flash CircuitPython (and other payloads). Thanks!

gbuzogany, 2020-07-23:
AFAIK you can't use NRF Connect to flash the software if you are using UF2 (it seems to only work with Open Bootloader).
Also, AFAIK, to use circuitpython, you should flash the Adafruit nRF52 bootloader:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases

The mdk_nrf52840_dongle_* should be what you are looking for.
After flashing the bootloader, you should be able to flash circuitpython (https://circuitpython.org/board/makerdiary_nrf52840_mdk_usb_dongle/)

I hope this is helpful.

AngainorDev, 2020-10-24:
Had the very same issue, that did it for me, details:
https://github.com/makerdiary/nrf52840-mdk-usb-dongle/issues/41