makerdiary/nrf52840-mdk-usb-dongle #37: ble 5.0 sniffer

Pillar1989, 2020-07-16:
@caizelin  @xiongyihui A BLE 5.0 sniffer is badly needed.

xiongyihui, 2020-07-16:
Try this one https://github.com/makerdiary/raccoon

Pillar1989, 2020-07-16:
@xiongyihui  support extend packets ?

gbuzogany, 2020-12-10:
I got the Nordic BLE sniffer working on the nrf52840-MDK, it works well with wireshark and seems to be very complete.
Download here:
https://www.nordicsemi.com/Software-and-tools/Development-Tools/nRF-Sniffer-for-Bluetooth-LE/Download#infotabs

Then follow the instructions here to get it working:
https://infocenter.nordicsemi.com/pdf/nRF_Sniffer_UG_v2.2.pdf

caizelin, 2020-12-11:
@gbuzogany Awesome!

closed 2020-12-14 by Pillar1989

gbuzogany, 2021-06-02:
It was on the MDK USB dongle (https://makerdiary.com/collections/frontpage/products/nrf52840-mdk-usb-dongle).

I used the sniffer_nrf52840dongle_nrf52840_7cc811f.hex

telethonic, 2021-06-02:
Thanks for the quick reply :-)

I'm getting closer, but not quite there yet. 

I wrote that firmware using nRF Connect,
put the extcap/* files in place,
and the snifffer shows up in wireshark.

When I start a capture though, wireshark just hangs :-|

Is there a way to check if the dongle was flashed correctly? 
Did you maybe use a different tool to flash it? (@gbuzogany )

gbuzogany, 2021-06-02:
I used a segger j-link to flash the board (SWD interface).
I use macOS.

Make sure to follow the installation guide:
https://infocenter.nordicsemi.com/pdf/nRF_Sniffer_UG_v2.2.pdf