makerdiary/nrf52840-mdk-usb-dongle #39: Reset Led is green

T2brozz, 2020-09-21:
Hello I am trying to get circuitpython on the dongle, but I am not able to do complete this guide. One problem is, taht now I don't have a red LED when I press reset and plug it in. Instead I have 2 green LEDS. Why is this happening ?

gbuzogany, 2020-10-06:
Which bootloader are you using? UF2? open bootloader?

T2brozz, 2020-10-06:
I want to flash ciruitpython on it and i bought it new. I used the instruction from ciruitpython.

gbuzogany, 2020-10-06:
The firmware that comes flashed to the board depends on when you bought it, and where you bought it from: Older boards come with open bootloader, newer boards with UF2.

If you hold the reset pin, and plug the board to the USB port of your computer, does a new "disk drive" appear on your computer? If yes, it's the UF2 bootloader.

To be able to run circuitpython on the board, you first need to flash the circuitpython bootloader for the nrf52840. There's a specific build for the mdk dongle (mdk_nrf52840_dongle_bootloader).
The latest release, today, can be found here:
https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases/tag/0.3.2

Instructions that can help:
https://learn.adafruit.com/circuitpython-on-the-nrf52/nrf52840-bootloader

T2brozz, 2020-10-06:
ok thank you, I will try it later. Are the  Instructions for the nrf52840 the same as for the nrf52840 mdk usb dongle?

gbuzogany, 2020-10-06:
They are not, and now I realise that the DK has a segger chip that you probably don't have. So these instructions (the ones from adafruit), won't help you much.

As you probably are using the UF2 bootloader (you should double check), check the instructions here:
https://github.com/adafruit/Adafruit_nRF52_Bootloader#making-your-own-uf2

Do you have a raspberry pi?

T2brozz, 2020-10-06:
Yeah I have a raspberry pi 4

gbuzogany, 2020-10-06:
In this case, maybe it's just simpler to flash the circuitpython bootloader using OpenOCD. There are pretty clear instructions in this video:
https://www.youtube.com/watch?v=R5wub5ywzTU

T2brozz, 2020-10-06:
Problem with this tutorial is that the makerdiary dongle has not open contacts. And also I have the one with the case and I cant open it . https://www.amazon.de/GeeekPi-nRF52840-MDK-USB-Dongle-Geh%C3%A4use/dp/B07TWRJRRC/ref=sxts_sxwds-bia-wc-p13n1_0?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=18W62NCM8SK02&cv_ct_cx=nrf52840+dongle&dchild=1&keywords=nrf52840+dongle&pd_rd_i=B07TWRJRRC&pd_rd_r=c712763b-9813-405a-aa4c-5e2983ca2974&pd_rd_w=339iW&pd_rd_wg=HWDCr&pf_rd_p=ba3091f1-24fd-4110-b1e7-0b0998d0dfa6&pf_rd_r=GGG4B7SWDZA11H0ZJGZB&psc=1&qid=1602002938&sprefix=nrf5%2Caps%2C160&sr=1-1-ac3a866f-801f-44fe-9e94-bb9a271cf6b8
 i bought this on

gbuzogany, 2020-10-06:
I understand.

It should be possible to open/close the case in a non-destructive way if you are careful, but anyway, you can try to follow the UF2 instructions here:
https://github.com/adafruit/Adafruit_nRF52_Bootloader#making-your-own-uf2

T2brozz, 2020-10-06:
Ok I will try it later

closed 2022-11-26 by T2brozz