makerdiary/nrf52840-mdk-usb-dongle #40: DFU bootloader does not seem to work

dh-metre, 2020-09-30:
When using nrf Connect and trying to connect to device there is an "Unsupported device. The detected device could not be recognized as neither JLink device nor Nordic USB device" error.  Found a webpage suggesting downloading and converting from hex to a new UF2 with updated bootloader but same issue.

When inserting drive with button pressed on startup the light flashes red and then goes to green.  This behavior does not seem to be described in documentation

gbuzogany, 2020-10-06:
AFAIK you can only use nRF Connect to program the device if you are using the open bootloader.
If you are using the UF2 bootloader, you should follow the docs to make the uf2 files, and then copy them to the drive.

shenyi0828, 2020-10-21:
> AFAIK you can only use nRF Connect to program the device if you are using the open bootloader.
> If you are using the UF2 bootloader, you should follow the docs to make the uf2 files, and then copy them to the drive.

How can I determine which bootloader I'm using? I've never changed since shipped.

Switchcrafter, 2020-10-30:
any update on this?   Having the same issue EXACTLY as @dh-metre described.  new dongle.

gbuzogany, 2020-10-30:
This page should explain how to switch between open bootloader and uf2 bootloader:
https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/

Another way is using openocd on a raspberry pi:
https://www.youtube.com/watch?v=R5wub5ywzTU

You could also use a J-Link programmer (or pretty much any other SWD programmer).