makerdiary/nrf52840-mdk-usb-dongle #41: Circuitpython install, wrong instructions and outdated doc - Up to date instructions

AngainorDev, 2020-10-24:
I have a nRF52840 MDK USB Dongle with case, with UF2 bootloader.

It enters bootloader (press button + insert, but with fixed green led, not blinking red)
bootloader info on udisk says UF2 Bootloader 0.2.13-42-g82464f9-dirty lib/nrfx (v2.0.0) lib/tinyusb (legacy-1500-g23df777b) s140 6.1.1

When trying to install circuitpython from https://circuitpython.org/board/makerdiary_nrf52840_mdk_usb_dongle/
copying uf2, reg light blinks, then device restarts and it's down.
No Udisk, no serial port, no device at all when issuing a "lsusb"
Tried with 5.3.1, 5.4.0, 6.0.0, no firmware works.
I can re-enter bootloader and re-copy the original uf2 I copied, there I get a serial port, but no other uf2 works.

Here is what worked for me (but should be made clear on the wiki)
- get the mdk_nrf52840_dongle_*.hex from  https://github.com/adafruit/Adafruit_nRF52_Bootloader/releases
- used uf2conv.py from https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/tools to convert to UF2 (please, provide bootloader as UF2 as well)
- enter bootloader (fixed green led), copy the converted uf2
- ends up with led red blinking regularly
- disconnect, re-enter bootloader (fixed green led)
- copy UF2 from https://circuitpython.org/board/makerdiary_nrf52840_mdk_usb_dongle/ , used the v6.0.0 UF2

Then the dongle reboots, with serial and circuitpython udisk

It's easy once you have all the - up to date - info at hand, please upgrade the doc!

Threadzless, 2020-12-05:
hey what arguments do you put into uf2conv.py to get a working .uf2? the one I create doesn't cause the dongle to reboot after i copy it over :(

AngainorDev, 2020-12-08:
I used
`uf2conv.py firmware.hex -c -f 0xADA52840`

Grazfather, 2021-01-22:
This worked for me.

Notably, from the releases linked, it now has zip files with uf2 already in them. *this did not work*. I had to manually convert the .hex to the .uf2 and flash that to get circuit python installed.

benjaTwose, 2021-04-17:
> 
> 
> This worked for me.
> 
> Notably, from the releases linked, it now has zip files with uf2 already in them. _this did not work_. I had to manually convert the .hex to the .uf2 and flash that to get circuit python installed.

Me too, thanks !!

giacomogallina, 2022-08-20:
Thank you so much, it worked perfectly!