makerdiary/nrf52840-mdk-usb-dongle #44: Vin and coin cell usage with the dongle

rananna, 2020-11-22:
The  pin out diagram suggests Vin should range dfom 3.3V ~ 5.5V , while the shematic diagram flags Vin as ranging from VIN:2.3~5.5V. Which one is correct?
I would like to power the 52840 dongle with a 3v lithium coin cell like the CR2032.
Is this possible?

caizelin, 2020-11-26:
Hi,
2.3~5.5V is the operating input range of the Step-Down Converter.  It's not recommended to supply 3V coin cell on Vin. A 3.7V battery will better.

rananna, 2020-11-26:
Thanks for the reply.
While waiting for an answer, I tried a 3V coin cell and it seemed to work fine. 
What is the dc-dc converter doing when there is a 3v input?
Since it is speced to work from 2.3v,  does it step up from a 3v vin to get 3.3v? 
What are the potential issues of using a coin cell on Vin?

rananna, 2020-11-26:
I don't want to use rechargeable batteries due to the inconvenience of charging and low mah capacity comparing to a standard cr2032.
Alternatively, could the dc-dc converter possibly work with 2 cr2032 cells in series at 6V?

rananna, 2020-11-26:
If the nrf52840 soc vdd can be powered by 3v, would it be better to bypass the DC-dc converter and connect the 3v coin cell directly to the vdd (3.3v) pin on the dongle?

gbuzogany, 2020-12-10:
All the limits for the nrf52 are available at the product specification doc:
https://infocenter.nordicsemi.com/pdf/nRF52840_PS_v1.1.pdf

5.5V seems to be the max input voltage at the VBUS pin.

The MDK board also has a step-down converter that also seem to have an upper limit of 5.5V.

You would need to use another DC-DC converter to drop the voltage within the range, but there will be a penalty in energy consumption, so choose well.

DaveVdE, 2025-01-22:
What about supplying 3V directly to the 3V3 pin? Would that annoy the step down converter?