makerdiary/nrf52840-mdk-usb-dongle #45: Unable to flash new code using Open Bootloader

mhbarnes, 2020-11-23:
Upon receiving the dongle, I attempted to switch from the pre-installed UF2 bootloader to Open bootloader, since I wanted to use the nrf connect tools. I was successful in doing so, though now I am unable to flash new code or upgrade back to the UF2 bootloader.  Whenever I attempt, I get the error "Timed out waiting for acknowledgement from the device" and "Attempting to use a port that is not open". Below is the output:

$ ls
README.md                                               nrf52840_usb_dongle_open_bootloader_v1_1_0.hex
nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.hex      nrf52840_usb_dongle_open_bootloader_v1_1_0.zip
nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip  uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip
nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.zip
$ nrfutil dfu serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p /dev/ttyS9
Upgrading target on /dev/ttyS9 with DFU package /home/mibarnes/Desktop/nrf/nrf52840-mdk-usb-dongle/firmware/open_bootloader/uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip. Flow control is disabled, Dual bank, Touch disabled
Timed out waiting for acknowledgement from device.

Failed to upgrade target. Error is: Attempting to use a port that is not open
Traceback (most recent call last):
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 294, in serial
    dfu.dfu_send_images()
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 232, in dfu_send_images
    self._dfu_send_image(HexType.BOOTLOADER, self.manifest.bootloader)
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 203, in _dfu_send_image
    self.dfu_transport.send_init_packet(init_packet)
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 155, in send_init_packet
    self.send_packet(packet)
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 241, in send_packet
    self.serial_port.write(bytearray(pkt.data))
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 531, in write
    raise portNotOpenError
serial.serialutil.SerialException: Attempting to use a port that is not open

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
$

I was able to successfully flash the blinky LED code to the dongle using the nrf programmer, but when trying to flash anything else, I get the following error:

20:29:14.352	DFU failed: Error message for known extended error code from DFU target: The init packet does not contain a signature. This bootloader requires DFU updates to be signed.
20:29:16.367	Device not found due to failure during DFU

I am using WSL for Windows 10. Is there any way to reset the bootloader that I'm missing? Thanks in advance.

caizelin, 2020-11-23:
Hi,
You may need to use the cmd like this(use `usb-serial` instead of `serial`):
```
nrfutil dfu usb-serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p <your-serial-port-name>
```

mhbarnes, 2020-11-23:
Doing so returns an error with nrfutil dfu:

```
$ ls
README.md                                               nrf52840_usb_dongle_open_bootloader_v1_1_0.hex
nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.hex      nrf52840_usb_dongle_open_bootloader_v1_1_0.zip
nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip  uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip
nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.zip
$ nrfutil dfu usb-serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p /dev/ttyS9
Usage: nrfutil dfu [OPTIONS] COMMAND [ARGS]...
Try "nrfutil dfu --help" for help.

Error: No such command "usb-serial".
$ nrfutil dfu --help
Usage: nrfutil dfu [OPTIONS] COMMAND [ARGS]...

  This set of commands support Nordic DFU OTA package generation for
  distribution to applications and serial DFU.

Options:
  --help  Show this message and exit.

Commands:
  genpkg  Generate a package for distribution to Apps supporting Nordic DFU
          OTA
  serial  Program a device with bootloader that support serial DFU
$
```

caizelin, 2020-11-24:
@mhbarnes It seems you are using the newer `nrfutil` tool.  The nrfutil version 5.0.0  works fine. Try to install the `nrfutil 5.0.0`:
```
pip uninstall nrfutil
pip install nrfutil==5.0.0
```

mhbarnes, 2020-11-24:
@caizelin I installed nrfutil 5.0.0, and tried installing the UF2 bootloader again. I hit a different error this time:

```
$ nrfutil dfu usb-serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p /dev/ttyS9

Traceback (most recent call last):
  File "/home/mibarnes/.local/bin/nrfutil", line 8, in <module>
    sys.exit(cli())
  File "/usr/lib/python3/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/lib/python3/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/lib/python3/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 975, in usb_serial
    do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, False)
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/__main__.py", line 941, in do_serial
    dfu.dfu_send_images()
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 125, in dfu_send_images
    self._dfu_send_image(self.manifest.bootloader)
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu.py", line 90, in _dfu_send_image
    self.dfu_transport.open()
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 213, in open
    self.__set_prn()
  File "/home/mibarnes/.local/lib/python3.8/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 307, in __set_prn
    self.dfu_adapter.send_message([DfuTransportSerial.OP_CODE['SetPRN']]
TypeError: can only concatenate list (not "map") to list
$
```

I ran into the same error when attempting to upload the blinky code as well.

gbuzogany, 2020-12-10:
Try with adafruit-nrfutil==0.5.3.post5

closed 2022-01-13 by mhbarnes