makerdiary/nrf52840-mdk-usb-dongle #46: Operating bootloader using ONLY usb interface

rananna, 2020-12-01:
Nordic provides utilities for DFU programming over USB,  and I would like to provide a custom bootloader to users to load using usb.  As far as I can tell the dongle’s existing bootloader can only be overwritten using an SWD compatible debugging probe such as Segger J-link. Is this the case? If not, how can I program a custom bootloader using only usb interface. 
I know I can change  to open bootloader but so far I cannot get my custom bootloader to program.
Any ideas?

caizelin, 2020-12-03:
Hi,
You can update your custom bootloader from UF2 Bootloader, as it does not need to sign your bootloader. Here is the steps you can follow:
https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/open_bootloader#change-to-open-bootloader-from-uf2-bootloader

You can find the source code of open bootloader here:
https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/examples/nrf5-sdk/open_bootloader

rananna, 2020-12-06:
You have outlined the process to change to open bootloader from the uf2 bootloader.
I want to replace the bootloader with a custom bootloader using only usb. Is this possible?

gbuzogany, 2020-12-10:
Yes it is possible, but you will probably need to implement your own.
https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.0.2/lib_bootloader_modules.html

chuckixia, 2021-06-07:
> Hi,
> You can update your custom bootloader from UF2 Bootloader, as it does not need to sign your bootloader. Here is the steps you can follow: [...]

this was exactly what I was looking for. Is it possible to put this on this website here:
https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/

also maybe make it clearer that the new dongles now ship with the new bootloader?