makerdiary/nrf52840-mdk-usb-dongle #49: nRF connect

andrewtholt, 2021-01-07:
Not an issue more of a question.

Will nRF connect for desktop work with this device ?

Thanks, should be a nice simple question for you .

Regards,
Andrew

Raboo, 2021-01-07:
Yes.

andrewtholt, 2021-01-09:
Hi,

Whilst your answer gets 10/10 for being concise it was not particularly helpful.

I have connected it to a Linux computer.

Started the nrf connect for desktop app. Clicked on "Bluetooth Low Energy" and the following errors:

Received status with code 0 PKT_SEND_MAX_RETRIES_REACHED, message: 'No response from device. Tried to send packet 6 times.'

serial port /dev/ttyACM0 closed.

Error occured when opening port. Errorcode: NRF_ERROR_TIMEOUT (0xd)

Do I need  specific firmware loaded into the dongle ? where would I find it ?

Thanks
Andrew

Raboo, 2021-01-11:
You have two bootloaders available (probably more).
Open Bootloader and U2F.
Open Bootloader was shipped with the devices previously, now I believe they might ship the device with U2F.
Open Bootloader is used with nRF Connect.

You can read a bit about it here https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/.
So you might have to switch to Open Bootloader if you want to use nRF Connect.

andrewtholt, 2021-01-13:
Hi,

Sorry to keep bothering you, Please see below.  I select a Hex, file
connect to device and I get. Any suggestions?

[image: image.png]

Raboo, 2021-01-13:
Just an FYI, I'm no expert, I just own a nrf52840-mdk-usb-dongle myself.

Anyways your image is not attached. Don't think attachments via email work on GitHub.

andrewtholt, 2021-01-13:
Hi,

Thanks for the response.

Pressing the reset, then powering doesn't put the device into DFU mode.
Any ideas about how I can recover it ?

Grazfather, 2021-01-22:
@andrewtholt , hold reset while plugging in the device, and hold the button until the second LED turns green.

You should now have a new disk mounted. You can put uf2 files there.

You probably want to put a uf2 for the OpenBootloader there. See #41 for the basic idea, but with circuit python.

andrewtholt, 2021-01-25:
Hi,

Thanks that cured my problem:

CURRENT.UF2
INDEX.HTM
INFO_UF2.TXT

I presume that CURRENT.UF2 is the bootloader

and INFO_UF2.TXT contains:

Model: nRF52840 MDK USB Dongle
Board-ID: nRF52840-Dongle-v1
Date: Feb  3 2020

Thanks again.
Andrew

chuckixia, 2021-06-07:
I would like to downgrade this device and use the nrf connect utility instead. How can go back to open boot loader?