makerdiary/nrf52840-mdk-usb-dongle #50: Zephyr support for this dongle version?

424778940z, 2021-01-08:
I see they only have support for full mdk but not this dongle version.
Any plan for it?

gbuzogany, 2021-01-18:
It's limited by hardware, as there's no programmer embedded on the board.
If you have a programmer (like a j-link) you can run zephyr on it like it was the full mdk.

424778940z, 2021-01-18:
Already made a PR for it
https://github.com/zephyrproject-rtos/zephyr/pull/31202

whjochim, 2021-05-11:
Apparently there is already support for this board.  I have already compiled and run the blinky example on it.

https://docs.zephyrproject.org/latest/boards/arm/nrf52840_mdk/doc/index.html
It originates from [this Pull Request in 2018](https://github.com/zephyrproject-rtos/zephyr/pull/10525) , which was merged

424778940z, 2021-05-11:
> https://docs.zephyrproject.org/latest/boards/arm/nrf52840_mdk/doc/index.html

dongle and normal version is not the same board

I got really busy recently, didn't get a chance to finish this pr

dchauran, 2021-05-12:
@424778940z I just commented on the zephyr PR, I would love to see this completed.