makerdiary/nrf52840-mdk-usb-dongle #51: Is there a BLE sniffer uf2 image for download directly?

belkinhappyzj, 2021-02-02:
My dongle purchased has a uf2 bootloader preinstalled,  nRF desktop doesn't directly work with it and I failed to upgrade the booloader to open bootloader either (serial port cannot open on bit 115200 for uknown reason, the serial port does show up though).
My goal is to use this dongle as a BLE sniffer, hence, can anyone provide a download of BLE sniffer uf2 image so that I can just copy it to the flash to make it work?  thank you.

mkotzjan, 2021-02-23:
Have a look at the [wiki](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/#dfu-via-uf2-bootloader), there is a [python conversion script](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/tools).

belkinhappyzj, 2021-02-23:
> Have a look at the [wiki](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/#dfu-via-uf2-bootloader), there is a [python conversion script](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/tools).

Got it, thank you

bennijamin, 2021-12-09:
> > Have a look at the [wiki](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/#dfu-via-uf2-bootloader), there is a [python conversion script](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/tools).
> 
> Got it, thank you

did it work for you? i try hard on it since yesterday and i cant get it to work. the blinky example works -> i'm able to flash the MDK example on it, but when i load the sniffer firmware, the LED blinks red and my PC doesnt show any connection to it. Wireshark isnt showing the device either.

SimonIT, 2023-02-06:
My problem was, that I firstly used the `sniffer_nrf52840dk_nrf52840_4.1.1.hex` file and not `sniffer_nrf52840dongle_nrf52840_4.1.1.hex`.
Inside the zip is my working uf2-file: [flash.zip](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/files/10627504/flash.zip)