makerdiary/nrf52840-mdk-usb-dongle #52: Green Solid LED When In Programming Mode

matt001k, 2021-02-11:
My LED stays green and both the nrfutil and NRF Connect programmer cannot find the chip. The NRF Connect app says the device is unsupported.
[image]

mkotzjan, 2021-02-23:
Same for me

mkotzjan, 2021-02-23:
Same for me. But it appears as a flash drive which, according to the [wiki](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/#dfu-via-uf2-bootloader), means that it's using the UF2 Bootloader

chuckixia, 2021-06-07:
this is happening for me too. I would like to use the original NRF Connect tool instead. How do I downgrade or change the bootloader to use the old one?

closed 2022-04-18 by matt001k