makerdiary/nrf52840-mdk-usb-dongle #55: nRF52840 MDK USB Dongle not connectable to nRF Connect

github-klas, 2021-03-09:
1. A new device not used before attached to computer with its button pressed to get in program mode (according to instruction on wiki.makerdiary.com).
2. The dongle's LED flashes red a few times and then turns steady green.
3. nRF Connect says "unsupported device" (please see attached screen dump).
What is wrong?

[Unsupported_Device.pdf](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/files/6108463/Unsupported_Device.pdf)

benjaTwose, 2021-04-17:
Exactly the same problem. Also, Trying circuitpython firmware uf2 fails. Also trying to change firmware with nrfutil fail.

michaelw3, 2021-05-07:
Same here tried to load a circuitpython uf2 file after trying to connect to the board in different ways.
tried on win 10 with nrf-connect got the same unsupported device message, tried on Linux to upload new fw same on win.
The red light keeps flashing but never turns green and the dongle doesn't appear as  Circuitpython.

thanks for any help with this issue

DamyHao, 2021-07-17:
Same here. Any solutions?

bo5972, 2022-11-23:
Hi!
I have the same problem, by chance did you manage to make it work?
Thanks :)

michaelw3, 2022-11-23:
> Hi! I have the same problem, by chance did you manage to make it work? Thanks :)

Hi,
I wasn't able to get it to work and basically gave up..
no help from their side and tried couple of different solutions that are scattered over the www