makerdiary/nrf52840-mdk-usb-dongle #56: AttributeError: 'dict' object has no attribute 'iteritems'

myrdrks, 2021-03-12:
Hey there.

When I try:
pi@raspberrypi:~/openthread/output/nrf52840/bin $ nrfutil pkg generate --hw-version 52 --sd-req 0x00 --application-version 1 --application  ot-rcp.hex app.zip

I get:
AttributeError: 'dict' object has no attribute 'iteritems'

When I then try to fix "/home/pi/.local/lib/python3.9/site-packages/nordicsemi/dfu/package.py:403":
-iteritems()
+items()
I get a lot more errors.

Please send help!

whjochim, 2021-05-11:
I just had the same Issue, it was solved for me by downgrading to Python 3.8 in an virtual environment. nrfutil doesn't seem to support Python 3.9 currently.
Hope it helps

phoebebright, 2022-11-21:
Am running python 3.9 and had same issue.  Followed these steps and started working (not sure why!)::

     sudo apt-get -y install libusb-1.0-0-dev sed
     pip3 install click crcmod ecdsa intelhex libusb1 piccata protobuf pyserial pyyaml tqdm pc_ble_driver_py
     pip3 install -U --no-dependencies nrfutil

zoechi, 2023-08-09:
I run into the same. 
https://github.com/makerdiary/nrf52840-mdk-usb-dongle/issues/56#issuecomment-1322301257 didn't help.  
Downgrading to Python 3.9 didn't help either because some links seemed to keep pointing to 3.11 and cause errors, that it's not available.  
I guess I need to purge everything and start over.

bjh45964, 2024-02-24:
> Am running python 3.9 and had same issue. Followed these steps and started working (not sure why!)::
>
> ```
>  sudo apt-get -y install libusb-1.0-0-dev sed
>  pip3 install click crcmod ecdsa intelhex [...]

This worked for me, fixed broken install of nrfutil (Was following https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.9.2-dev1/matter/openthread_rcp_nrf_dongle.html), working on Raspbian 11.1, nrfutil-6.1.7, Python 3.9.2, pip 20.3.4