makerdiary/nrf52840-mdk-usb-dongle #58: I'm sick of this

gfmoore, 2021-05-23:
I've bought a nrf52840-mdk-usb-dongle from China.
I've downloaded nrf connect onto a windows machine and a linux machine.
I've tried inserting into usb and inserting into usb with reset pin held down.

On Linux nrfconnect|Programming I get message : 
Unsupported device. The detected device could not be recognized as neither JLink device nor Nordic USB device.

On Windows I get message : 
Error when calling version command: Error: Error message: Timeout while reading from serial transport. See https://github.com/NordicSemiconductor/pc-nrfconnect-launcher/blob/master/doc/serial-timeout-troubleshoot.md
15:08:50.694	Error when fetching device versions: TypeError: Cannot read property 'part' of undefined

I've spent days trying to follow instructions for installing wpantund at https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/getting-started/ and gave up because it is too flippin complicated. I only want to make a flashing light not land a rover on Mars....

I've then looked at other solutions such as using nrfUtil. I can't get anywhere because the instructions suddenly assume knowledge that I haven't got or don't know how to resolve and no amount of searching on google reveals an answer that helps.

For instance I thought maybe if I load the open bootloader I might get somewhere
nrfutil dfu usb-serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p <your-serial-port-name>

How the hell do I find the <your-serial-port-name>?????

I tried /dev/ttyACM0 which seemed reasonable, no joy just more incomprehensible errors.

So please tell me I've bought a knock off piece of junk (though it does what flash red a few times etc) or I'm doing something so ridiculously foolish that I deserve to be flamed.

So please tell me what to do to get let's say nRF Connect running on Windows for this device. Now that can't be hard can it?

Gordon, a really hacked off and fed up Gordon ...

gfmoore, 2021-05-23:
I'm now trying nrfutil on windows - man that was also a battle to install.

So trying to flash the openbootloader
nrfutil dfu usb-serial -pkg uf2_bootloader-0.2.13-44-gb2b4284-nosd_signed.zip -p com4
 as per YOUR DOC

Guess what another error:

Usage: nrfutil dfu [OPTIONS] COMMAND
Try 'nrfutil dfu --help' for help.

Error: No such command 'usb-serial'.

I mean apart from the com4 bit I am just copying the text.....

gfmoore, 2021-05-23:
So changed usb-serial to serial

Now getting a whole heap of messages : 

PS C:\Users\Gordon\Downloads\nrf52840-mdk-usb-dongle-master\firmware\open_bootloader> nrfutil dfu serial -pkg uf2_bootloader-0.2.13-44-gb2
4-nosd_signed.zip -p com4
>>
Upgrading target on com4 with DFU package C:\Users\Gordon\Downloads\nrf52840-mdk-usb-dongle-master\firmware\open_bootloader\uf2_bootloader
.13-44-gb2b4284-nosd_signed.zip. Flow control is disabled, Dual bank, Touch disabled

Failed to upgrade target. Error is: Serial port could not be opened on com4. Reason: could not open port 'com4': FileNotFoundError(2, 'The
tem cannot find the file specified.', None, 2)
Traceback (most recent call last):
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu_transport_serial.py", line 113, in open
    self.serial_port = Serial(port=self.com_port, baudrate=self.baud_rate, rtscts=self.flow_control, timeout=self.timeout)
  File "c:\python39\lib\site-packages\serial\serialwin32.py", line 33, in __init__
    super(Serial, self).__init__(*args, **kwargs)
  File "c:\python39\lib\site-packages\serial\serialutil.py", line 244, in __init__
    self.open()
  File "c:\python39\lib\site-packages\serial\serialwin32.py", line 64, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'com4': FileNotFoundError(2, 'The system cannot find the file specified.', None, 2)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\__main__.py", line 294, in serial
    dfu.dfu_send_images()
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu.py", line 232, in dfu_send_images
    self._dfu_send_image(HexType.BOOTLOADER, self.manifest.bootloader)
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu.py", line 157, in _dfu_send_image
    self.dfu_transport.open()
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu_transport_serial.py", line 115, in open
    raise NordicSemiException("Serial port could not be opened on {0}. Reason: {1}".format(self.com_port, e))
nordicsemi.exceptions.NordicSemiException: Serial port could not be opened on com4. Reason: could not open port 'com4': FileNotFoundError(
The system cannot find the file specified.', None, 2)

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
PS C:\Users\Gordon\Downloads\nrf52840-mdk-usb-dongle-master\firmware\open_bootloader>

I'll try and change the com4 parameters in device manager

man oh man.

gfmoore, 2021-05-23:
Changed baud rate

PS C:\Users\Gordon\Downloads\nrf52840-mdk-usb-dongle-master\firmware\open_bootloader> nrfutil dfu serial -pkg uf2_bootloader-0.2.13-44-gb2b4
4-nosd_signed.zip -p com5
>>
Upgrading target on com5 with DFU package C:\Users\Gordon\Downloads\nrf52840-mdk-usb-dongle-master\firmware\open_bootloader\uf2_bootloader-0
.13-44-gb2b4284-nosd_signed.zip. Flow control is disabled, Dual bank, Touch disabled

Failed to upgrade target. Error is: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))
Traceback (most recent call last):
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\__main__.py", line 294, in serial
    dfu.dfu_send_images()
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu.py", line 232, in dfu_send_images
    self._dfu_send_image(HexType.BOOTLOADER, self.manifest.bootloader)
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu.py", line 206, in _dfu_send_image
    self.dfu_transport.send_firmware(firmware)
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu_transport_serial.py", line 213, in send_firmware
    self.send_packet(pkt)
  File "C:\Users\Gordon\AppData\Roaming\Python\Python39\site-packages\nordicsemi\dfu\dfu_transport_serial.py", line 241, in send_packet
    self.serial_port.write(bytearray(pkt.data))
  File "c:\python39\lib\site-packages\serial\serialwin32.py", line 317, in write
    raise SerialException("WriteFile failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: WriteFile failed (PermissionError(13, 'The device does not recognize the command.', None, 22))

Possible causes:
- Selected Bootloader version does not match the one on Bluefruit device.
    Please upgrade the Bootloader or select correct version in Tools->Bootloader.
- Baud rate must be 115200, Flow control must be off.
- Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
PS C:\Users\Gordon\Downloads\nrf52840-mdk-usb-dongle-master\firmware\open_bootloader>

I noted:
 Target is not in DFU mode. Ground DFU pin and RESET and release both to enter DFU mode.
How can I do that on a usb dongle, I thought that was the point.

gfmoore, 2021-05-23:
So just to make clear:
I take the device, hold down the reset button, insert it into a usb port.
The top left led (at the usb port is green, the other led flashes red a couple of times and then goes green. I am in DFU mode right?

gfmoore, 2021-05-23:
I can now convert hex files to uf2 files. So tried blinky, but can't see any led blinking every 500ms - I assume that is what it should do. 

nrfConnect |programmer still doesn't recognise the device. Lost count of the things I've tried, but nrfutil on windows didn't work under Python. I eventually found a Nordic windows executable that does what it should, but not with my device. I've now ordered some proper nrf dongles and this is going in the bin. A waste of my money, a waste of my life and simple a waste.

Tara

closed 2021-05-23 by gfmoore

JRyuzaki, 2021-09-15:
Are there any updates on this issue? I am currently having the same problem.

gfmoore, 2021-09-16:
Sorry, no. I'd forgotten all about it. I later tried the nrf52840 dongle and dk board and though I got some things working, it turned out to be just as  much a nightmare as this thing. If you look on the Nordic site you will find some of my posts and unhappiness :sigh. I do have a word document that explains how I got VS Code set up on Windows for the nrf 52840 if you want a copy, though it's not fully edited. My biggest block was when I found that without seggers and stuff I didn't understand I found I couldn't get buttonless? OTA to work on the dongle. I am about to dip my feet into BT again, but dreading it. 
Gordon