makerdiary/nrf52840-mdk-usb-dongle #60: Cannot enter programming mode!!

madaeon01, 2021-07-07:
I bought one nrf52840 mdk usb dongle.
If I simply insert it into USB, the dongle is seen as DFU removable disk, I see the 3 files onboard including current.uf2. Written a blinky test with Arduino IDE and Adafruit nrf52 libraries, converted the .hex to uf2 with uf2conv.py . **But getting the dongle to connect into programming mode** (the red led blinks), which is supposed to happen if you press reset when inserting the dongle, **almost never happens**. Otherwise the led is green, you still see the DFU removable disk, but trying to copy/replace files only causes the device to disconnect. On two PCs (one has windows 7, the other windows 10), tried with USB3 and USB2 ports, I was able to upload the new UF2 file ONCE. All the other times, on Windows 10 if red led is still blinking after you release it, the USB device is not recognized anymore. With green light, the device is visible, but can't copy files on it. On Windows 7 notebook, I was able to upload the files once. But after that time, the red light will never stay on, when the device is inserted by pressing the reset button. It always turns green after a while. Looking at https://github.com/makerdiary/nrf52840-mdk-usb-dongle/issues/new it seems other people have this issue, and no solution is available??

DamyHao, 2021-07-18:
The documentation needs to be updated and improved a lot. This worked for me (linux):

1- Download the bootloader from [https://github.com/makerdiary/nrf52840-mdk-usb-dongle/raw/master/firmware/open_bootloader/nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip](url)
2- Install nrfutil using `pip install nrfutil`
3- Put the open_bootloader into the device using `nrfutil dfu serial -pkg /home/YOUR_USERNAME/Downloads/nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip -p /dev/ttyACM0`

Tiebe, 2022-03-07:
I keep getting the error `TypeError: __init__() got an unexpected keyword argument 'dfu_version'`
I have tried different versions of python (2.7, 3.9 and 3.10)

This is the full stack trace:
```
Traceback (most recent call last):
File "/home/tiebe111/.local/bin/nrfutil", line 33, in <module>
sys.exit(load_entry_point('nrfutil==6.1.3', 'console_scripts', 'nrfutil')())
File "/home/tiebe111/.local/lib/python3.9/site-packages/click/core.py", line 1128, in __call__
return self.main(*args, **kwargs)
File "/home/tiebe111/.local/lib/python3.9/site-packages/click/core.py", line 1053, in main
rv = self.invoke(ctx)
File "/home/tiebe111/.local/lib/python3.9/site-packages/click/core.py", line 1659, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/home/tiebe111/.local/lib/python3.9/site-packages/click/core.py", line 1659, in invoke
return _process_result(sub_ctx.command.invoke(sub_ctx))
File "/home/tiebe111/.local/lib/python3.9/site-packages/click/core.py", line 1395, in invoke
return ctx.invoke(self.callback, **ctx.params)
File "/home/tiebe111/.local/lib/python3.9/site-packages/click/core.py", line 754, in invoke
return __callback(*args, **kwargs)
File "/home/tiebe111/.local/lib/python3.9/site-packages/nordicsemi/__main__.py", line 1063, in serial
do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, True,
File "/home/tiebe111/.local/lib/python3.9/site-packages/nordicsemi/__main__.py", line 972, in do_serial
dfu = Dfu(zip_file_path = package, dfu_transport = serial_backend, connect_delay = connect_delay)
File "/home/tiebe111/.local/lib/python3.9/site-packages/nordicsemi/dfu/dfu.py", line 69, in __init__
self.manifest           = Package.unpack_package(zip_file_path, self.unpacked_zip_path)
File "/home/tiebe111/.local/lib/python3.9/site-packages/nordicsemi/dfu/package.py", line 652, in unpack_package
return Manifest.from_json(_json)
File "/home/tiebe111/.local/lib/python3.9/site-packages/nordicsemi/dfu/manifest.py", line 200, in from_json
return Manifest(**kwargs['manifest'])
TypeError: __init__() got an unexpected keyword argument 'dfu_version'
```