makerdiary/nrf52840-mdk-usb-dongle #62: Dongle never enters DFU mode. LED always green

L13w, 2021-10-30:
After plugging in the dongle without holding down the user switch, I see a solid green light in the bottom right side of the dongle. The device shows as COM5, but times out when trying to connect with nRF connect. When inserting the device with the button down, there is anohter led on the mid-left side of the board that blinks red momentarily and then a solid green color (two green LEDs). It never turns red, never enters DFU mode.

ks00x, 2022-01-01:
I was also confused by this and I think the documentation is to blame. It looks like the nRF connect app does not work with this device, at least not with the original bootloader. However you can progam it via simply copying over the uf2 firmware file to the device!  When you press the button while inserting the device, a disk drive will appear in windows explorer. There is a conversion script for hex->uf2 firmware conversion but I found the BT LE sniffer firmware I needed on this [Adafruit](https://learn.adafruit.com/ble-sniffer-with-nrf52840?view=all) site.

vigilancer, 2022-03-18:
Or you could flash Open Bootloader and then DFU will work as documented.

https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/open_bootloader

Take a look at last answer:
https://devzone.nordicsemi.com/f/nordic-q-a/71617/problems-burning-firmware-into-nrf52840-mdk-usb-dongle

ReX1983, 2023-04-24:
> Or you could flash Open Bootloader and then DFU will work as documented.
> 
> https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/open_bootloader
> 
> Take a look at last answer: https://devzone.nordicsemi.com/f/nordic-q-a/71617/problems-burning-firmware-into-nrf52840-mdk-usb-dongle

I flashed the Open Bootloader but now the device is not recognized by nRF Connect / nRF Programmer.
I have also tried jlink and nrfutil, but neither works. 

The dongle shows as nRF52 SDFU USB under Ports, in Windows.
 
Irrespective if I press or not the reset button when I insert the dongle, the red led pulses (it is not blinking but pulsing). Is this expected? I am not sure if it stuck in DFU mode or if I can't enter it.

I can't even go back to the UF2 bootloader, since nrfutil doesn't work.