makerdiary/nrf52840-mdk-usb-dongle #64: HandleRcpTimeout() at radio_spinel_impl.hpp:2226: RadioSpinelNoResponse AND [CRIT]-PLAT----: Init() at hdlc_interface.cpp:151: InvalidArgument

h4k1337, 2022-02-07:
Hei People, 

I want to install OpenThread on the MDK dongle, sure, I read the tutorial of course, I noticed that the build is outdated in the official documentation, that brings some problems for me.

I have the latest version from https://github.com/openthread/borderrouter
The installation looks fine, the Agent + Web Agent is running, but the dongle is not connected. The only information I can get from the syslog is:

With the Firemware from: https://github.com/makerdiary/nrf52840-mdk-usb-dongle/blob/master/firmware/openthread/ncp/thread_ncp_ftd_nrf52840_mdk_usb_dongle_v1.3.0.uf2

>b  7 17:15:02 raspberrypi systemd[1]: Starting OpenThread Border Router Agent...
Feb  7 17:15:03 raspberrypi kernel: [1142792.261706] usb 1-1.5: new full-speed USB device number 4 using dwc_otg
Feb  7 17:15:03 raspberrypi kernel: [1142792.425093] usb 1-1.5: config 1 has an invalid interface number: 2 but max is 1
Feb  7 17:15:03 raspberrypi kernel: [1142792.425120] usb 1-1.5: config 1 has no interface number 0
Feb  7 17:15:03 raspberrypi kernel: [1142792.427957] usb 1-1.5: New USB device found, idVendor=1915, idProduct=cafe, bcdDevice= 1.00
Feb  7 17:15:03 raspberrypi kernel: [1142792.427975] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Feb  7 17:15:03 raspberrypi kernel: [1142792.427990] usb 1-1.5: Product: nRF52840 MDK USB Dongle
Feb  7 17:15:03 raspberrypi kernel: [1142792.428005] usb 1-1.5: Manufacturer: Makerdiary
Feb  7 17:15:03 raspberrypi kernel: [1142792.428019] usb 1-1.5: SerialNumber: C8A0B95A308D
Feb  7 17:15:03 raspberrypi mtp-probe: checking bus 1, device 4: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
Feb  7 17:15:03 raspberrypi mtp-probe: bus: 1, device: 4 was not an MTP device
Feb  7 17:15:03 raspberrypi kernel: [1142792.496073] cdc_acm 1-1.5:1.1: ttyACM0: USB ACM device
Feb  7 17:15:03 raspberrypi kernel: [1142792.497169] usbcore: registered new interface driver cdc_acm
Feb  7 17:15:03 raspberrypi kernel: [1142792.497187] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
Feb  7 17:15:03 raspberrypi mtp-probe: checking bus 1, device 4: "/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5"
Feb  7 17:15:03 raspberrypi mtp-probe: bus: 1, device: 4 was not an MTP device
Feb  7 17:15:04 raspberrypi kernel: [1142793.241776] Under-voltage detected! (0x00050005)
Feb  7 17:15:07 raspberrypi systemd[1]: Started OpenThread Border Router Agent.
Feb  7 17:15:07 raspberrypi otbr-agent[13741]: [INFO]-UTILS---: Running 0.3.0-1fc5673a2
Feb  7 17:15:07 raspberrypi otbr-agent[13741]: [INFO]-UTILS---: Thread version: 1.2.0
Feb  7 17:15:07 raspberrypi otbr-agent[13741]: [INFO]-UTILS---: Thread interface: wpan0
Feb  7 17:15:07 raspberrypi otbr-agent[13741]: [INFO]-UTILS---: Backbone interface: wlan0
Feb  7 17:15:07 raspberrypi otbr-agent[13741]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Feb  7 17:15:07 raspberrypi otbr-agent[13741]: [INFO]-UTILS---: Radio URL: trel://wlan0
Feb  7 17:15:08 raspberrypi kernel: [1142797.401776] Voltage normalised (0x00000000)
Feb  7 17:15:09 raspberrypi otbr-agent[13741]: 62d.22:29:25.749 [CRIT]-PLAT----: HandleRcpTimeout() at radio_spinel_impl.hpp:2226: RadioSpinelNoResponse
Feb  7 17:15:09 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Feb  7 17:15:09 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Feb  7 17:15:14 raspberrypi systemd[1]: otbr-agent.service: Service RestartSec=5s expired, scheduling restart.
Feb  7 17:15:14 raspberrypi systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 134442.
Feb  7 17:15:14 raspberrypi systemd[1]: Stopped OpenThread Border Router Agent.
Feb  7 17:15:14 raspberrypi systemd[1]: Starting OpenThread Border Router Agent...
Feb  7 17:15:18 raspberrypi systemd[1]: Started OpenThread Border Router Agent.
Feb  7 17:15:18 raspberrypi otbr-agent[13764]: [INFO]-UTILS---: Running 0.3.0-1fc5673a2
Feb  7 17:15:18 raspberrypi otbr-agent[13764]: [INFO]-UTILS---: Thread version: 1.2.0
Feb  7 17:15:18 raspberrypi otbr-agent[13764]: [INFO]-UTILS---: Thread interface: wpan0
Feb  7 17:15:18 raspberrypi otbr-agent[13764]: [INFO]-UTILS---: Backbone interface: wlan0
Feb  7 17:15:18 raspberrypi otbr-agent[13764]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Feb  7 17:15:18 raspberrypi otbr-agent[13764]: [INFO]-UTILS---: Radio URL: trel://wlan0
Feb  7 17:15:20 raspberrypi otbr-agent[13764]: 62d.22:29:36.330 [CRIT]-PLAT----: HandleRcpTimeout() at radio_spinel_impl.hpp:2226: RadioSpinelNoResponse
Feb  7 17:15:20 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Feb  7 17:15:20 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Feb  7 17:15:25 raspberrypi systemd[1]: otbr-agent.service: Service RestartSec=5s expired, scheduling restart.
Feb  7 17:15:25 raspberrypi systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 134443.
Feb  7 17:15:25 raspberrypi systemd[1]: Stopped OpenThread Border Router Agent.
Feb  7 17:15:25 raspberrypi systemd[1]: Starting OpenThread Border Router Agent...
Feb  7 17:15:28 raspberrypi systemd[1]: Started OpenThread Border Router Agent.
Feb  7 17:15:28 raspberrypi otbr-agent[13787]: [INFO]-UTILS---: Running 0.3.0-1fc5673a2
Feb  7 17:15:28 raspberrypi otbr-agent[13787]: [INFO]-UTILS---: Thread version: 1.2.0
Feb  7 17:15:28 raspberrypi otbr-agent[13787]: [INFO]-UTILS---: Thread interface: wpan0
Feb  7 17:15:28 raspberrypi otbr-agent[13787]: [INFO]-UTILS---: Backbone interface: wlan0
Feb  7 17:15:28 raspberrypi otbr-agent[13787]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Feb  7 17:15:28 raspberrypi otbr-agent[13787]: [INFO]-UTILS---: Radio URL: trel://wlan0
Feb  7 17:15:30 raspberrypi otbr-agent[13787]: 62d.22:29:46.778 [CRIT]-PLAT----: HandleRcpTimeout() at radio_spinel_impl.hpp:2226: RadioSpinelNoResponse
Feb  7 17:15:30 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=6/NOTCONFIGURED
Feb  7 17:15:30 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.

Of course I tried to flash the last version from the repository
Then I came up with the idea and wanted to build the firmware myself, from: https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/getting-started/#update-ncp-firmware

> Feb  7 17:00:08 raspberrypi systemd[1]: Starting OpenThread Border Router Agent...
[... 40 lines cut ...]
Feb  7 17:00:34 raspberrypi systemd[1]: Stopped OpenThread Border Router Agent.
Feb  7 17:00:34 raspberrypi systemd[1]: Starting OpenThread Border Router Agent...
Feb  7 17:00:37 raspberrypi systemd[1]: Started OpenThread Border Router Agent.
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: [INFO]-UTILS---: Running 0.3.0-1fc5673a2
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: [INFO]-UTILS---: Thread version: 1.2.0
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: [INFO]-UTILS---: Thread interface: wpan0
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: [INFO]-UTILS---: Backbone interface: wlan0
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: [INFO]-UTILS---: Radio URL: spinel+hdlc+uart:///dev/ttyACM0
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: [INFO]-UTILS---: Radio URL: trel://wlan0
Feb  7 17:00:37 raspberrypi otbr-agent[11308]: 62d.22:14:53.777 [CRIT]-PLAT----: Init() at hdlc_interface.cpp:151: InvalidArgument
Feb  7 17:00:37 raspberrypi systemd[1]: otbr-agent.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
Feb  7 17:00:37 raspberrypi systemd[1]: otbr-agent.service: Failed with result 'exit-code'.
Feb  7 17:00:42 raspberrypi systemd[1]: otbr-agent.service: Service RestartSec=5s expired, scheduling restart.
Feb  7 17:00:42 raspberrypi systemd[1]: otbr-agent.service: Scheduled restart job, restart counter is at 134340.
Feb  7 17:00:42 raspberrypi systemd[1]: Stopped OpenThread Border Router Agent.
Feb  7 17:00:42 raspberrypi systemd[1]: Starting OpenThread Border Router Agent...

and so on... 

After a "big-little" searching, it turned out that there is now a separate Github project for the dongle. But I didn't get any further with that.
https://github.com/openthread/ot-nrf528xx

Can someone please help me, I'm slowly running out of ideas, and I don't want the dongle to end up as a mail giver (paperweight).

Thanks in Advance

ameerhazo, 2022-08-11:
Hi @h4k1337, I've been dealing with the same problem. Have you managed to find the solution for this problem?

Regards,
Ameer

philb32, 2023-02-11:
I've added a new issue which might help : https://github.com/makerdiary/nrf52840-mdk-usb-dongle/issues/83