makerdiary/nrf52840-mdk-usb-dongle #74: ubuntu 22.04 can not transfer the UF2 zip file into the USB drive

Teymurian, 2022-08-29:
Ubuntu 22.04 can not transfer the UF2 zip file into the USB drive.
after pushing the reset button and connecting it to the USB port, then trying to copy the uf2 file it shows this error
PermissionError: [Errno 13] Permission denied: '/dev/ttyACM0'
which is fixed by
sudo chmod 666 /dev/ttyACM0
next try can't finish the copy process and this is the error: 

> nrfutil pkg generate --hw-version 52 --sd-req 0x00 --application-version 1 --application ~/Downloads/adafruit-circuitpython-makerdiary_nrf52840_mdk_usb_dongle-en_US-7.3.2.hex  app_dfu_pkg_test2.zip
> |===============================================================|
> |You are not providing a signature key, which means the DFU     |
> |files will not be signed, and are vulnerable to tampering.     |
> |This is only compatible with a signature-less bootloader and is|
> |not suitable for production environments.                      |
> |===============================================================|
> Zip created at app_dfu_pkg_test2.zip
> sudo chmod 666 /dev/ttyACM0
> sudo ls -l  /dev/ttyACM0
> crw-rw-rw- 1 root dialout 166, 0 Aug 29 00:10 /dev/ttyACM0
>
> nrfutil dfu usb-serial -pkg app_dfu_pkg_test2.zip -p /dev/ttyACM0 
>   [------------------------------------]    0%2022-08-29 00:11:36,459 LIBUSB_ERROR_ACCESS: Unable to connect to trigger interface.
> 
> Traceback (most recent call last):
>   File "/home/ed/.local/bin/nrfutil", line 8, in <module>
>     sys.exit(cli())
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1128, in __call__
>     return self.main(*args, **kwargs)
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1053, in main
>     rv = self.invoke(ctx)
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1659, in invoke
>     return _process_result(sub_ctx.command.invoke(sub_ctx))
>   File "/usr/lib/python3/dist-packages/click/core.py", line 1395, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/usr/lib/python3/dist-packages/click/core.py", line 754, in invoke
>     return __callback(*args, **kwargs)
>   File "/home/ed/.local/lib/python3.10/site-packages/nordicsemi/__main__.py", line 1032, in usb_serial
>     do_serial(package, port, connect_delay, flow_control, packet_receipt_notification, baud_rate, serial_number, False,
>   File "/home/ed/.local/lib/python3.10/site-packages/nordicsemi/__main__.py", line 988, in do_serial
>     dfu.dfu_send_images()
>   File "/home/ed/.local/lib/python3.10/site-packages/nordicsemi/dfu/dfu.py", line 127, in dfu_send_images
>     self._dfu_send_image(self.manifest.application)
>   File "/home/ed/.local/lib/python3.10/site-packages/nordicsemi/dfu/dfu.py", line 88, in _dfu_send_image
>     self.dfu_transport.open()
>   File "/home/ed/.local/lib/python3.10/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 217, in open
>     self.__get_mtu()
>   File "/home/ed/.local/lib/python3.10/site-packages/nordicsemi/dfu/dfu_transport_serial.py", line 366, in __get_mtu
>     self.mtu = struct.unpack('<H', bytearray(response))[0]
> TypeError: cannot convert 'NoneType' object to bytearray

could you help me with this?

AndrewBarfield, 2023-07-14:
This is a long shot... but any update on this?

caizelin, 2023-08-09:
@AndrewBarfield Could you try the latest [UF2 Bootloader 0.7.1](https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/main/firmware/uf2_bootloader/0.7.1)

Xiao-Ming-r, 2023-12-25:
Did you find a solution?