makerdiary/nrf52840-mdk-usb-dongle #75: Programming nrf52840

titou4307, 2022-10-01:
Hello
I have bought the nrf52840 under enclosure USB

I want to program this with the "nrfconnect-setup-3.12.0-ia32.exe" but impossible to access to this peripheral ....

I have error message : 

18:19:06.675	Unsupported device. The detected device could not be recognized as neither JLink device nor Nordic USB device.

Maybe anyone has a solution ?

EnriqueSV, 2022-11-14:
Hello,
Can you fix the problem? I've the same issue,
Tank you

titou4307, 2022-11-14:
Hi
Yes ....After many and many try with the software NrfUtil, NrfUtil-adafruit and the soft Nrf52840 of the constructor

But it is not a perfect science !! and I cannot reproduce the how to....

Of memory I have use the soft of the constructor for upload various firmware....And after more upload I could go into download mode for continue the process

EnriqueSV, 2022-11-25:
Hi finally I can Change to Open Bootloader from UF2 Bootloader.  In this web explain the steps https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/open_bootloader. If you use Windows or Mac  you can download the adafruit-nrfutil in this link https://github.com/adafruit/Adafruit_nRF52_nrfutil/releases
In windows:
1- put in the same folder the adafruit-nrfutil.exe and the nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip,
2- press the dongle button and connect to PC
3- Search the COMX of the USB

and execute this command: 
adafruit-nrfutil --verbose dfu serial --package nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip -p COM19 -b 115200 --singlebank
[image]
[image]
I hope this help