makerdiary/nrf52840-mdk-usb-dongle #79: cant flash nrf52 dongle using nRFconnect for desktop.

sohaibbinaltaf, 2022-11-13:
I am using nRF52840 dongle with raspberry pi 4 to create a thread network. I downloaded nRFconnect for desktop to flash the hex files into the dongles. 

I successfully flashed two dongles, but today when i was trying to flash the third dongle i was unable to write the hex file. The write option was disabled and could only be used when i select the Enable MCUboot. Not sure why. 

When i select Enable MCUboot and try to write the hex file it shows an error, 
mcu boot failed with error operation, program not supported error when starting fw program

can you please help me to solve this problem.

sohaibbinaltaf, 2022-11-13:
i am trying to flash ot-cli-ftd.hex into the dongle

sohaibbinaltaf, 2022-11-14:
how can i flash the dongle from the raspberry pi command line, what packages i need to install and what commands will be used if i could get this information.

mkotzjan, 2023-02-10:
Hope you managed to flash it by now, but otherwise please refer to the [wiki](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/)