makerdiary/nrf52840-mdk-usb-dongle #83: Using nrf52840 MDK dongle as RCP in openthread border router

philb32, 2023-02-11:
Just FYI I have made a bit of progress with this, using the dongle on a raspberry pi running the opnthread border router software with the dongle acting as an RCP as opposed to an NCP as this seems to be the way to go. 

Following the instructions to use the dongle as an NCP working with wpantund almost work, but it seems like the NCP software is resetting itself every few seconds and there is no simple way to try and debug it.

As I didn't manage to build an image that worked with the UF2 bootloader, I switched over to the OpenBootloader, instructions for doing this (UF2 -> OpenBootloader and OpenBootloader -> UF2) are in the MDK here : 

https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/master/firmware/open_bootloader

These instructions work and I switched back and forth between UF2 and OpenBootloader a couple of times without issue. When you switch to OpenBootloader this means that you can use the nRF Connect Programmer tool to flash the dongle, it doesn't work with the UF2 bootloader installed.

Now you can build the Radio Coprocessor sample from the nRF Connect SDK 2.2.0 from the command line following the instructions on this page :

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/protocols/thread/tools.html#configuring-a-radio-co-processor

Note, to get a command line with a suitable environment for running west etc., start  nrF Connect for Desktop, open Toolchain Manager, and then on the dropdown list to the right of nRF Connect SDK v2.2.0 select Open Terminal, and this will open a terminal with the evironment set correctly. I found that the coprocessor sample wouldn't build from the VS Code extension, so just do it from the command line as per the instructions.

The resulting hex file is zephyr.hex in ~/ncs/2.2.0/build/zephyr (I installed the SDK immediately under my home directory). You can also look at the zephyr.map file in that directory and you can see the symbol rom_start at offset 0x0000000000001000 which is encouraging because we need to avoid 0x0 - 0xfff. So now, plug the dongle into your build machine, run nRF Connect Programmer (from nRF Connect for Desktop), select the device (top left), drag and drop the zephyr.hex file to the drag and drop window, and click the Write button on the left hand side to flash the device. After writing, there is some kind of error as the board immediately reboots to the new image - you can ignore this.

On the same web page, after the section on configuring a radio coprocessor, there is a short section on how to setup a raspberry pi to run OTBR, i.e. follow the official open thread codelab ( https://openthread.io/codelabs/openthread-border-router#1 ) with some minor changes.

Once the raspberry pi is set up, you can plug the nRF52840 dongle into it and get going with the rest of it. 

For an actual thread device, I've got an nRF5340 development board from Nordic. However I haven't got as far as checking whether the radio on the nRF52840 MDK dongle is actually working yet. 

This is as far as I got (log from raspberry pi with dongle), all looks very encouraging : 

`pi@raspberrypi:~ $ 
pi@raspberrypi:~ $ sudo service otbr-agent status
● otbr-agent.service - OpenThread Border Router Agent
   Loaded: loaded (/lib/systemd/system/otbr-agent.service; enabled; vendor preset: enabled)
   Active: active (running) since Fri 2023-02-10 17:49:14 GMT; 1min 21s ago
  Process: 646 ExecStartPre=/usr/sbin/service mdns start (code=exited, status=0/SUCCESS)
 Main PID: 668 (otbr-agent)
    Tasks: 1 (limit: 2062)
   CGroup: /system.slice/otbr-agent.service
           └─668 /usr/sbin/otbr-agent -I wpan0 -B wlan0 spinel+hdlc+uart:///dev/ttyACM0 trel://wlan0

Feb 10 17:49:14 raspberrypi otbr-agent[668]: 00:00:00.065 [I] Platform------: Execute command `ipset swap otbr-ingress-deny-src-swap otbr-ingr
Feb 10 17:49:14 raspberrypi otbr-agent[668]: 00:00:00.072 [I] Platform------: Execute command `ipset swap otbr-ingress-allow-dst-swap otbr-ing
Feb 10 17:49:14 raspberrypi otbr-agent[668]: 00:00:00.072 [I] Platform------: MulticastRoutingManager: Disable: OK
Feb 10 17:49:14 raspberrypi otbr-agent[668]: [INFO]-BA------: Publish meshcop service OpenThread BorderRouter #AA5A._meshcop._udp.local.
Feb 10 17:49:14 raspberrypi otbr-agent[668]: 00:00:00.072 [I] RouterTable---: Route table
Feb 10 17:49:14 raspberrypi otbr-agent[668]: 00:00:00.075 [I] Platform------: [netif] Host netif is down
Feb 10 17:49:15 raspberrypi otbr-agent[668]: [INFO]-MDNS----: Received reply for service OpenThread BorderRouter #AA5A._meshcop._udp., service
Feb 10 17:49:15 raspberrypi otbr-agent[668]: [INFO]-MDNS----: Successfully registered service OpenThread BorderRouter #AA5A._meshcop._udp.
Feb 10 17:49:15 raspberrypi otbr-agent[668]: [INFO]-BA------: Result of publish meshcop service OpenThread BorderRouter #AA5A._meshcop._udp.lo
Feb 10 17:49:15 raspberrypi otbr-agent[668]: [INFO]-BA------: Result of publish meshcop service OpenThread BorderRouter #AA5A._meshcop._udp.lo
pi@raspberrypi:~ $ sudo ot-ctl dataset init new
Done
pi@raspberrypi:~ $ sudo ot-ctl dataset commit active
Done
pi@raspberrypi:~ $ sudo ot-ctl ifconfig up
Done
pi@raspberrypi:~ $ sudo ot-ctl thread start
Done
pi@raspberrypi:~ $ sudo ot-ctl state
leader
Done
pi@raspberrypi:~ $ sudo ot-ctl netdata show
Prefixes:
fd1a:ea75:7318:1::/64 paos low b000
Routes:
fd1a:ea75:7318:2:0:0::/96 sn low b000
fd17:b73b:c9b6:290d::/64 s med b000
Services:
44970 01 1c000500000e10 s b000
44970 5d fd387531c92339572ae66a080129d8d9d11f s b000
Done
pi@raspberrypi:~ $ sudo ot-ctl ipaddr
fd38:7531:c923:3957:0:ff:fe00:fc11
fd38:7531:c923:3957:0:ff:fe00:fc10
fd38:7531:c923:3957:0:ff:fe00:fc38
fd1a:ea75:7318:1:77:f1bc:c520:fe2e
fd38:7531:c923:3957:0:ff:fe00:fc00
fd38:7531:c923:3957:0:ff:fe00:b000
fd38:7531:c923:3957:2ae6:6a08:129:d8d9
fe80:0:0:0:44ec:fd72:d477:aa5a
Done
pi@raspberrypi:~ $ sudo ot-ctl dataset active -x
0e080000000000010000000300001735060004001fffe0020817b73bc9b661290d0708fd387531c9233957051026c3133603908afcbf9e9c7ebed84589030f4f70656e5468726561642d3066323501020f2504104a5e9aca158d90f771905ee33f19a7200c0402a0f7f8
Done
pi@raspberrypi:~ $ 
`
For help with joining device to the network and controlling the device, you can get the Nordic Android CHIPTool app from here : 

https://github.com/nrfconnect/sdk-connectedhomeip/releases

olavt, 2024-01-17:
I used the recipe above using the nRF52840 MDK USB Dongle w/ Case, and it looks like the OpenThread Border Router starts and works as expected. I can form a new network and get the Thread dataset required for pairing a device. The pairing process with a Matter over Thread device however fails to complete successfully. I have now tried it a few times (after forming a new network from scratch) and with the nrf52840 MDK dongle as an RCP the pairing of a Matter over Thread device fails every time.

If I use this dongle (built from the nRF Connect SDK) it works fine.

https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dongle

I also notice that after copying the ot-rcp.uf2 to the Makerdiary nRF52840 MDK USB Dongle w/ Case, I'm no longer able to set it into programming mode. 

Is that expected?

studiofuga, 2024-01-17:
No, it's not expected. I haven't tested it, since I usually use a programmer. Unfortunately I can't test the pairing at the moment (I have no matter over thread device right now). Sorry.

olavt, 2024-01-17:
@studiofuga How can I recover to get into programming mode? I have the Dongle w/ Case. Can it be opened easily without destroying it?