makerdiary/nrf52840-mdk-usb-dongle #106: DAPLink source for nrf52840-mdk-usb-dongle

fanoush, 2024-09-22:
Hello, I'd like to rebuild https://github.com/makerdiary/nrf52840-mdk-usb-dongle/blob/main/firmware/daplink/daplink-nrf52840_mdk_usb_dongle-v0258.uf2 from source

Found your fork https://github.com/makerdiary/DAPLink but it only contains commit for your Pitaya-Link board. Can you make your patches for  nrf52840 MDK dongle available?

Also looks like  the pitaya board was merged upstream https://github.com/ARMmbed/DAPLink/pull/896 but can't find MDK dongle upstream.

fanoush, 2024-09-30:
Well I found at least this PR with some other nrf52840 devices https://github.com/ARMmbed/DAPLink/pull/1015 - Nordic 52840 dongle and XIAO BLE, no nrf52840-mdk-usb-dongle there though.

h0m3us3r, 2024-11-24:
Second this

rkyurpan256, 2025-02-14:
@caizelin is the source for nrf52840-mdk-usb-dongle's DAPLink firmware closed and thus not provided? If not, could it be shared, much appreciated.

rkyurpan256, 2025-02-27:
Hello @fanoush and @h0m3us3r  , you can check again https://github.com/ARMmbed/DAPLink/pull/1015 . We have been working to add support for Makerdiary's nrf52840-mdk-usb-dongle. You can check it out and test it how it works. I am already using it in manufacturing process for our flash programmers. And now you can build the DAPLink firmware by yourself :)