makerdiary/nrf52840-mdk-usb-dongle #114: Issue with my Makerdiary Nrf52840 dongle : Not able to change the UF2 firmware

ylg2935, 2025-02-14:
### Discussed in https://github.com/makerdiary/nrf52840-mdk-usb-dongle/discussions/113

<div type='discussions-op-text'>

<sup>Originally posted by **ylg2935** February 14, 2025</sup>
Hello ,
I just bought a GeeekPi nRF52840 Micro Dev Kit USB Dongle on Amazon 
i'm following the guidelines indicated for the dongle  ( https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/)
pressing the button, I can do a Drag/Drop operation on Windows and Linux and put the new firmware, but on both, the new  Firwmare file that I put (nrf802154_sniffer_v0.7.2.uf2) is never saved (the new FW is lost after reboot, I'm retrieving the old one).
I have successed to download an open bootloader (the traces is below) but still the same result, I'm not able to put the firmware  with the 802.15.4 analyzer...
Could you please help me ?
My goal is to do a 802.15.4 analyzer with this  nRF52840  dongle. 
Do you know how I can flash the nrf802154_sniffer_v0.7.2.uf2 using adafruit-nrfutil ? there are not providing the full package (I mean the uf2.zip file).

_C:\nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0>adafruit-nrfutil --verbose dfu serial --package nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip -p com3
Upgrading target on com3 with DFU package C:\nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0\nrf52840_mdk_usb_dongle_open_bootloader_v1.2.0.uf2.zip. Flow control is disabled, Dual bank, Touch disabled
Opened serial port com3
Starting DFU upgrade of type 2, SoftDevice size: 0, bootloader size: 34544, application size: 0
Sending DFU start packet
Sending DFU init packet
Sending firmware file
########################################
############################
Activating new firmware

DFU upgrade took 29.97208333015442s
Device programmed._

Regards,

Yvon LG</div>

mobcom, 2025-04-13:
Having the same issue.

philipandwaffle, 2025-04-20:
Having the same issue, bought the devkit about a year ago and only getting round to playing with it now. Looks like my bootloader is on version 0.3.0-25-gb3d9a0a. The [documentation](https://wiki.makerdiary.com/nrf52840-mdk-usb-dongle/programming/uf2boot/#updating-the-uf2-bootloader) seems to suggest I should at least be on be on version 4.0.0  by default, not sure if that's the issue.

After dragging and dropping the latest version the INFO_UF2.txt file doesn't change to match the version I dragged into it.

[image]

wizmo64, 2025-06-02:
Same issue.  Tried the published file at https://github.com/makerdiary/nrf52840-mdk-usb-dongle/tree/main/firmware/ble_sniffer and it does nothing.  Tried converting current official version from Nordic obtained during nrfutil setup and using the uf2conv utility; at least that shows blinking red to confirm flashing.  On power cycle I get only green light, no USB device.  None of the other uf2 files from the main/firmware collection work, not even the original ble_connectivity.  Too bad I did not come here first to see reported issues and no response from makerdiary.

ivan-ushakov, 2025-06-25:
@wizmo64 could you please tell me what exactly you did? When I use `west` to build my project I see that address is 0x1000:

```
[233/233] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      184048 B       844 KB     21.30%
             RAM:       42353 B       256 KB     16.16%
        IDT_LIST:          0 GB         2 KB      0.00%
Converting to uf2, output size: 368128, start address: 0x1000
Wrote 368128 bytes to zephyr.uf2
```